CauldronDevelopmentLLC/CAMotics #232: Bullnose tool support

jcoffland, 2017-05-04:
Tool shape terminology differs.  Here we are referring toroidal shaped tools where the doughnut hole is filled in.  In this case, the tool has no concave parts and is therefore far less complicated to simulate.  This is related to #231.

[image]
[image]

Mayov17, 2025-01-05:
Hello, any chance to use Bull Nose tool ?

jcoffland, 2025-01-06:
Not anytime soon unless someone wants to pay for it.

Mayov17, 2025-01-06:
how much for this add ?

jcoffland, 2025-01-06:
It would not be cheap.  You email me directly if you are serious about opening a contract for such a project.  joseph@cauldrondevelopment.com