CauldronDevelopmentLLC/CAMotics #233: Install on Debian 7.11 fails

DougCoffland, 2017-05-12:
I tried to install CAMotics on my new computer in my garage, but was not successful.  The package installer did not work at all.  I was able to compile from source, but the resulting package does not install complaining about dependencies.  It says libv8-3.14.5 is not installed when I run dpkg.  I tried installing libv8-3.14.5 and was successful, but dpkg still fails on the camotics package that I built.

I can start camotics from the command line, but I immediately get an error in the console that says: "ERROR:Exception: String format '%.*f' invalid".  Then I can't do much from there.

jcoffland, 2017-05-12:
What was the error from the package installer?  Can you post it here?

DougCoffland, 2017-05-12:
I decided to remove camotics and cbang and start over completely.  Here's
what I experienced.

When I clone cbang, I get the following warning:
"WARNING: gnome-keyring:: couldn't connect to:
/home/doug/.cache/keyring-DW6Mge/pkcs11: No such file or directory"
Then cbang continues to clone successfully.

When I clone camotics, I get the same warning as I got when cloning cbang.
Then it continues to clone successfully.

scons -C cbang completes successfully with no errors or warnings.

export CBANG_HOME=PWD/cbang is successful.

cd CAMotics
scons
Completes successfully with no errors or warnings.

At this point, I attempted to run camotics from the command line.  It
opens, but I get the following error:
"ERROR:Exception: String format '%.*f' invalid"
The default program (camotics) does not run.

I tried loading another program that I wrote and it fails too, giving the
following error:
ERROR:Exception: @:2:8 Expected one of 'NnTtFf-.0123456789\"[{' but found
','

scons package completes successfully

sudo dpkg -i camotics_*.deb results in the following error:
dpkg: dependencyt problems prevent configuration of camotics
 camotics depends on libv8-3.14.5; however:
  Package libv83.15.5 is not installed

jcoffland, 2017-05-12:
It sounds like there are several different problems here:

  1. Did you have a problem installing the regular package from the Website?  What has it?
  2. C! has a number formatting bug.  I've fixed this.  Update your C! and rebuild both.
  3. There is an unsatisfied dependency on ``libv8-3.14.5``.

For the last item.  Does the package install work if you run this first?:

    sudo apt-get update
    sudo apt-get install -y libv8-dev

DougCoffland, 2017-05-14:
I did have a problem installing the regular package.  I did not remember
exactly what it was so I went back and removed the current camotics and
then did a "sudo gdepi camotics_*.deb" from my download folder, which is
where I downloaded it in the first place.  I got the following error:

Dependency is not safisfiable: libqt5core5a

As recommended, I did a pull on cbang and reran scons.  That appears to
have  been successful.

I did the sudo apt-get update and sudo apt-get install -y libv8-dev
commands and it said "libV8-dev is already the newest version."

I decided to go back and completely remove cbang and CAMotics folders and
then clone them once again.

After doing that cbang built with no errors or warnings.

CAMotics compiled with no errors or warnings, but when I built the package
it started off with the following warning:

scons: warning: QT5DIR variable not defined, and detected moc is for Qt
4.8.2
File "/home/doug/CAMotics/config/qt5/__init__.py, line 401, in _detect

Then, it gave another warning

scons: warning: QT4DIR variable is not defined, using moc executable as a
hint (QT4DIR=/usr)
File "/home/doug/CAMotics/config/qt4/__init__.py, line 394, in _detect

Then it went on to complete building the package.

Finally, when I ran "sudo dpkg -i camotics_*.deb", I got the same complaint
saying that Package libv8-3.14.5 is not installed.

Doug

jcoffland, 2017-05-16:
So on the phone you said this worked:

    sudo dpkg -i --force-depends camotics_*.deb

Looks like we still need to sort out the package dependencies.  The initial install problem was because your system does not support Qt5 which is a requirement for those packages.  The system is just too old.  The compile issue was fixed.

DougCoffland, 2017-05-16:
Unfortunately Linux CNC does not run on a newer version of linux.  This
will be another advantage for our box but will be a bit limiting for users
of camotics.

jcoffland, 2017-05-17:
I'm closing this because the libv8 issue is already documented in #215

closed 2017-05-17 by jcoffland