CauldronDevelopmentLLC/CAMotics #234: Fail to start Linux .deb file 1.1.1

kd7ura, 2017-05-17:


kd7ura, 2017-05-17:
sudo apt-get distribution-upgrade didnt fix GLIBCXX_3.4.20 error
run  
/home $ strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX
gets 3.4.19
so  digdug around and found

sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt-get install gcc-4.9
sudo apt-get upgrade libstdc++6
RUN  /home $ strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX
gets
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
:+1:

jcoffland, 2017-05-17:
This is a duplicate of #178.  The libv8 issue is documented in #215.  README.md issue solved with referenced commit.

kd7ura, 2017-05-17:
camotics: symbol lookup error: camotics: undefined symbol: _ZN7QString13toUtf8_helperERKS_

?????

kd7ura, 2017-05-17:
still trying to get it running Linux mint 17.1 qt5.2 installed currently upgrading to 17.3 
Its not just #178 , and #215,  its a failure to run 1.1.1 and having deb installer show all dependencies are met before install ,and getting dependencies errors when trying to start the program. im doing what i can to get it running but digging for dependencies that should have been covered by the  .deb is what the issue is about, im not psychic, cant read the dependencies from the either i rely on the deb doing what its supposed to do, like list all missing dependencies it cant find or showing the statement that dependencies cant be met or are missing Dependencies cannot install.

jcoffland, 2017-05-17:
The package is also not psychic.  I does not know what dependencies your particular Linux distro needs and it is nearly impossible to create packages that cover all the cases.  This is a problem with release binaries packages for Linux outside of the distro's packaging system.

Please provide more information about how you got ``camotics: symbol lookup error: camotics: undefined symbol: ZN7QString13toUtf8_helperERKS``.  I need to narrow this issue down to one solvable problem.  If there are other problems that are not cover by this issue or other issues then please open a new issue.  And please provide enough context so that I can reproduce the problem.

kd7ura, 2017-05-17:
1 one i downloaded the linux64.deb ver 1.1.1.
2 clicked it, package installer starts shows all dependencies are met,  click install button, installs camotics
click icon in other gets mouse timer 5/10 seconds then nothing.
so terminal camotics gets v8 error, i fix, terminal again camotics gets   camotics: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version GLIBCXX_3.4.20' not found
so upgradei upgrade /libstdc++.so.6 to lates.
terminal again  camotics  gets the QT error not shure yet what is about but gots to be QT.camotics: symbol lookup error: camotics: undefined symbol: ZN7QString13toUtf8_helperERKS
i was lookin through the .md for reqs,  will be checking mine  as soon as upgrade is finished.  
im running a supper fast 1.3 MB frontier communications dsl hookup back in a bit

kd7ura, 2017-05-18:
here is a paste of term.

 ~ $ camotics
*********************************** CAMotics ***********************************
     Website: http://camotics.org/
   Copyright: (c) 2011-2015
      Author: Joseph Coffland <joseph@camotics.org>
Organization: Cauldron Development LLC
        Args: 
      Config: <none>
************************************ Build *************************************
     Version: 1.1.1
        Date: Feb 26 2017
        Time: 07:03:12
  Repository: Git
    Revision: 0349ce6eb88d1e63c77bc7a50dd5e0653036a16d
      Branch: master
    Compiler: GNU 4.9.2
     Options: -std=c++11 -Wno-deprecated-declarations -O3 -funroll-loops -fPIC
    Platform: linux2 4.9.0-1-amd64
        Bits: 64
        Mode: Release
************************************ System ************************************
         CPU: Intel(R) Celeron(R) CPU 900 @ 2.20GHz
      CPU ID: GenuineIntel Family 6 Model 23 Stepping 10
        CPUs: 1
      Memory: 1.89GiB
 Free Memory: 226.72MiB
     Threads: POSIX_THREADS
  OS Version: 3.13
 Has Battery: true
  On Battery: false
  UTC Offset: -7
         PID: 2489
         CWD: /home/home
         DPI: 96
********************************************************************************
camotics: symbol lookup error: camotics: undefined symbol: _ZN7QString13toUtf8_helperERKS_
home@home-Presario-CQ56-Notebook-PC ~ $ 

should be a method for dgkg to run script at install of deb to check system for depends, or when building deb provide the libs for the build in additional folder in the build directory and link to these on install and run, the package builder knows what the package needs or should, he built it . no psy skills need then.
package should script start run dpkg,check deps ,install deps, check again till all package deps are met
or run a package deps check befor building the deb and include them in the deb file.
we shouldnt need to install all the latest software development and debug packages and IDEs to run the program , as for the error above i believe its just another lib.
i checked readme.sd for 
## Prerequisites
   - C!         - http://cbang.org/
-  - Qt4        - http://qt-project.org/
-  - Cairo2     - http://cairographics.org/
+  - Qt5        - http://qt-project.org/
   - SCons      - http://www.scons.org/
-  - ChakraCore - https://github.com/Microsoft/ChakraCore/
+  - v8         - https://developers.google.com/v8/

i upgradeded to 17.3 and not seeing , libqt5websockets-dev,   i have libqt5opengl5-dev but not the listed libqt5opengl-dev, (scons) not installed ,some cairo installed ,ChakraCore nothing found.
so the statment  in the deb installer, Status All Dependencies are satisfied is false.
im digin around try and find a solution for the dpkg check

kd7ura, 2017-05-18:
- more, added curl,
ran the curl build.sh it added 42 more dependencies listed below, and
biggest problem is   No package 'Qt5WebSockets' found,    not available for older versions of ubuntu

Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
zlib1g-dev is already the newest version.
zlib1g-dev set to manually installed.
libv8-dev is already the newest version.
libqt4-dev is already the newest version.
libqt4-dev set to manually installed.
scons is already the newest version.
The following extra packages will be installed:
  dpkg-dev g++ g++-4.8 git-man icu-devtools libboost-filesystem1.54-dev
  libboost-filesystem1.54.0 libboost-iostreams1.54-dev libboost-regex1.54-dev
  libboost-regex1.54.0 libboost-system1.54-dev libboost1.54-dev
  libcairo-script-interpreter2 liberror-perl libfontconfig1-dev
  libfreetype6-dev libglib2.0-dev libice-dev libicu-dev libpcre3-dev
  libpcrecpp0 libpixman-1-dev libpng12-dev libqt4-sql-sqlite libsm-dev
  libstdc++-4.8-dev libxcb-shm0-dev libxrender-dev x11proto-render-dev
Suggested packages:
  debian-keyring g++-multilib g++-4.8-multilib gcc-4.8-doc libstdc++6-4.8-dbg
  git-daemon-run git-daemon-sysvinit git-doc git-el git-email git-gui gitk
  gitweb git-arch git-bzr git-cvs git-mediawiki git-svn libboost1.54-doc
  libboost-atomic1.54-dev libboost-chrono1.54-dev libboost-context1.54-dev
  libboost-coroutine.54-dev libboost-date-time1.54-dev
  libboost-exception1.54-dev libboost-graph1.54-dev
  libboost-graph-parallel1.54-dev libboost-locale1.54-dev libboost-log.54-dev
  libboost-math1.54-dev libboost-mpi1.54-dev libboost-mpi-python1.54-dev
  libboost-program-options1.54-dev libboost-python1.54-dev
  libboost-random1.54-dev libboost-serialization1.54-dev
  libboost-signals1.54-dev libboost-test1.54-dev libboost-thread1.54-dev
  libboost-timer1.54-dev libboost-wave1.54-dev libboost1.54-tools-dev
  libmpfrc++-dev libntl-dev libcairo2-doc libglib2.0-doc libice-doc icu-doc
  libsm-doc sqlite3-doc libstdc++-4.8-doc qt4-doc-html
Recommended packages:
  libalgorithm-merge-perl libssl-doc qt4-designer qt4-doc
The following NEW packages will be installed:
  build-essential dpkg-dev g++ g++-4.8 git git-man icu-devtools
  libboost-filesystem-dev libboost-filesystem1.54-dev
  libboost-filesystem1.54.0 libboost-iostreams-dev libboost-iostreams1.54-dev
  libboost-regex-dev libboost-regex1.54-dev libboost-regex1.54.0
  libboost-system-dev libboost-system1.54-dev libboost1.54-dev libbz2-dev
  libcairo-script-interpreter2 libcairo2-dev liberror-perl libexpat1-dev
  libfontconfig1-dev libfreetype6-dev libglib2.0-dev libice-dev libicu-dev
  libpcre3-dev libpcrecpp0 libpixman-1-dev libpng12-dev libqt4-opengl-dev
  libqt4-sql-sqlite libsm-dev libsqlite3-dev libssl-dev libstdc++-4.8-dev
  libxcb-shm0-dev libxrender-dev qt4-dev-tools x11proto-render-dev
0 upgraded, 42 newly installed, 0 to remove and 4 not upgraded.

ran scons: warning: QT5DIR variable is not defined, using moc executable as a hint (QT5DIR=/usr)
File "/tmp/CAMotics/config/qt5/__init__.py", line 406, in _detect
  Compiler: gcc (gnu)
  Platform: posix
      Mode: gnu
      Arch: x86
      Jobs: 1
Checking for C library pthread... yes
Checking for C header file zlib.h... yes
Checking for C library z... yes
Checking for C header file bzlib.h... yes
Checking for C library bz2... yes
Checking for C header file expat.h... yes
Checking for C library expat... yes
Checking for C header file pthread.h... yes
Checking for C library pthread... yes
Checking for C library dl... yes
Checking for C library sqlite3... yes
[... 52 lines cut ...]
  Repository: Git
    Revision: 0349ce6eb88d1e63c77bc7a50dd5e0653036a16d
      Branch: master
    Compiler: GNU 4.9.2
     Options: -std=c++11 -Wno-deprecated-declarations -O3 -funroll-loops -fPIC
    Platform: linux2 4.9.0-1-amd64
        Bits: 64
        Mode: Release
************************************ System ************************************
         CPU: Intel(R) Celeron(R) CPU 900 @ 2.20GHz
      CPU ID: GenuineIntel Family 6 Model 23 Stepping 10
        CPUs: 1
      Memory: 1.89GiB
 Free Memory: 285.71MiB
     Threads: POSIX_THREADS
  OS Version: 3.13
 Has Battery: true
  On Battery: false
  UTC Offset: -7
         PID: 12025
         CWD: /home/home
         DPI: 96
********************************************************************************
camotics: symbol lookup error: camotics: undefined symbol: _ZN7QString13toUtf8_helperERKS_
home@home-Presario-CQ56-Notebook-P

jcoffland, 2019-01-19:
The remaining issue is better described in #263.

closed 2019-01-19 by jcoffland