CauldronDevelopmentLLC/CAMotics #235: Consider adding AppStream metadata

srakitnican, 2017-06-01:
https://www.freedesktop.org/software/appstream/docs/

> AppStream provides the foundation to build software-center applications. It additionally provides specifications for things like an unified software metadata database, screenshot services and various other things needed to create user-friendly application-centers for software distributions.

I've created some basic file for package which can be enhanced further.
```xml
<?xml version="1.0" encoding="UTF-8"?>
<component type="desktop-application">
  <id>CAMotics.desktop</id>
  <metadata_license>CC0-1.0</metadata_license>
  <project_license>GPL-2.0+</project_license>
  <name>CAMotics</name>
  <summary>Open-Source Simulation &amp; Computer Aided Machining - A 3-axis CNC GCode simulator</summary>
  <description>
    <p>
      CAMotics is an Open-Source software which can simulate
      3-axis NC machining. It is a fast, flexible and user friendly simulation
      software for the DIY and Open-Source community.
    </p>
    <p>
      At home manufacturing is one of the next big technology revolutions. Much like
      the PC was 30 years ago. There have been major advances in desktop 3D printing
      (e.g.  Maker Bot) yet uptake of desktop CNCs has lagged despite the
      availability of cheap CNC machines. One of the major reasons for this is a
      lack of Open-Source simulation and CAM (model to tool path conversion)
      software. CAM and NC machine simulation present some very difficult
      programming problems as evidenced by 30+ years of academic papers on these
      topics. Whereas 3D printing simulation and tool path generation is much
      easier. However, such software is essential to using a CNC.
    </p>
      Being able to simulate is a critical part of creating usable CNC tool paths.
      Programming a CNC with out a simulator is cutting with out measuring; it's
      both dangerous and expensive. With CAMotics you can preview the results of
      your cutting operations before you fire up your machine. This will
      save you time and money and open up a world of creative possibilities by
      allowing you to rapidly visualize and improve upon designs without wasting
      material or breaking tools.
    <p>
    </p>
  </description>
  <screenshots>
    <screenshot type="default">
      <image>http://camotics.org/images/screenshots/camotics.png</image>
    </screenshot>
    <screenshot>
      <image>http://camotics.org/images/screenshots/simulation_view.png</image>
    </screenshot>
    <screenshot>
      <image>http://camotics.org/images/screenshots/tool_view.png</image>
    </screenshot>
    <screenshot>
      <image>http://camotics.org/images/screenshots/tool_path_view.png</image>
    </screenshot>
    <screenshot>
      <image>http://camotics.org/images/screenshots/workpiece_view.png</image>
    </screenshot>
    <screenshot>
      <image>http://camotics.org/images/screenshots/full_layout.png</image>
    </screenshot>
  </screenshots>
  <url type="homepage">http://camotics.org/</url>
</component>
```
If I've got it right file should be named the same as `.desktop` file with `.appdata.xml` sufix. So I've named it `CAMotics.appdata.xml`.

jcoffland, 2017-06-01:
Can you submit this as a PR?

srakitnican, 2017-06-01:
My first pull request hopefully I've got it right. #236 
I've changed component type to "desktop" in order to be compatible with older AppStream tools. At least RHEL7 is still using those.
>The desktop-application component type is the same as the desktop component type - desktop is the older type identifier for desktop-applications and should not be used for new metainfo files, unless compatibility with very old AppStream tools (pre 2016) is still wanted.

https://www.freedesktop.org/software/appstream/docs/sect-Metadata-Application.html

jcoffland, 2017-06-01:
You got it right.  Thanks.!

closed 2017-06-01 by jcoffland