CauldronDevelopmentLLC/CAMotics #244: Add explicit support for LASER cutters

jcoffland, 2017-08-07:
Some people are already using CAMotics to view LASER and 3D printer tool paths.  A special mode for LASERs would view from the top by default and disable the cut workpiece simulation.

arthurlutz, 2019-04-19:
I am one of such users, and CAMotics does already a great job, so thanks for that ! 

One of the things I can't get it to do, is use it to check raster type laser engraving. The way the gcode is generated is that it goes back and forth and the laser has varying intensity so having something like https://github.com/CauldronDevelopmentLLC/CAMotics/issues/184 for that would be awesome so one could check the gcode is correct. I use https://gitlab.com/PlateformeC/laserengraver_smoothie/ (which uses raster2gcode) and can provide a gcode example file if needed.

jcoffland, 2019-04-19:
The latest CAMotics already does this if the intensity is varied using S words.  Just click the LASER icon on the toolbar to enable it.  Does your code vary LASER intensity using something else?

An example:

[image]

Without LASER mode it looks like this:

[image]

sk8nfool, 2020-01-15:
You ask if my laser cutter uses something else that S parameter to control intensity. The answer is yes! My Cutter uses a "fan" code to control laser intensity. Basically it turns the laser on and off with M106 and M107 codes without parameters. (Camotics complains about both of these.) M106 and M107 in Marlin have a number of default parameters which Camotics doesn't seem to understand.

jcoffland, 2020-01-16:
@sk8nfool I've added #318 for this.