CauldronDevelopmentLLC/CAMotics #245: CATCH_WARNING() and throws function related error

badrik91, 2017-08-10:
Hello everyone
i have a doubt in catch_warning and throws function .In this, i got error like invalid operands of types ‘const char [11]’ and ‘std::vector<std::__cxx11::basic_string<char> >::size_type {aka long unsigned int}’ to binary ‘operator<<’ this and expected primary-expression before ‘)’ token.
please give solution i am in new for this project.

jcoffland, 2017-08-10:
You need to provide a lot more information here.  Please tell us your OS, the version you installed, or code you're building, what instructions you're following and the exact error including line numbers.  Otherwise, it's nearly impossible to diagnose the problem you've encountered.

badrik91, 2017-08-11:
I am using ubuntu 16.04 OS,QT 5.9 and i follow this  link https://github.com/CauldronDevelopmentLLC/CAMotics and build in linux ,follow the instuction for linux.I installed all dependency which is given in instructionThis is the exactly error .
[image]
please give answer.
 thanks for your reply .

badrik91, 2017-08-11:
I have define this kind of macro then like this,
[image] is it right or wrong?
then get this kind of error like,
[image]

jcoffland, 2017-08-12:
You cannot build CAMotics from Qt creator.  The instructions tell you how to do it from the command line using ``scons``.  See: http://camotics.org/download.html#source-code or https://github.com/CauldronDevelopmentLLC/CAMotics#building-from-source

closed 2017-08-12 by jcoffland

badrik91, 2017-08-13:
Thank you sir for giving reply.
But I did this using commandline also but it simply run this in terminal.
but i want to build this in framework like Qt,Visual studio.but you said
not to build in Qt then where I build this project?
Please give solutions.

Closed #245 <https://github.com/CauldronDevelopmentLLC/CAMotics/issues/245>.

jcoffland, 2017-08-13:
You have to build it on the command line using ``scons``.  Follow the instructions I linked above.  If you encounter an error then paste it here.

badrik91, 2017-08-17:
Hello Sir,
I followed the link you have given but it create direct exe but i want to
modify the code for my requirement.so how i get successfully compiled ?
please give solution .

jcoffland, 2017-08-17:
If you want to build from source you must follow the directions I linked to above.

badrik91, 2017-08-17:
Hello Sir,
I have a problem in  related scon which is given bellow
[image]
please give solution .
I followed all the steps which you are given .

jcoffland, 2017-08-17:
Remove the directory ``/home/spr/CAMotics/camotics``.  It is preventing the build from working.

badrik91, 2017-08-17:
How can i remove this?because i have installed source code in /home/spr/CAMotics.so what can i do?
please give solution

jcoffland, 2017-08-17:
My recommendation is to delete everything in your ``/home/spr/CAMotics`` directory and start over.  If you follow the [directions](https://github.com/CauldronDevelopmentLLC/CAMotics#building-from-source) **precisely**, it does work.  Many other people have compiled CAMotics successfully.

badrik91, 2017-08-18:
Thank you sir.