CauldronDevelopmentLLC/CAMotics #246: Problem arise in cbang build in Windows.

badrik91, 2017-08-21:
Hello Sir,
I have problem in cbang build in windows .I have installed  all package in windows which you given in github .but i face error like this,
[image].
So how can i solve ? is it require new package for this?
Please give solution .

jcoffland, 2017-08-21:
I made a change that might fix this problem.  Please update your cbang and rebuild both cbang and CAMotics.

If that does not work, since you are getting an error on line 37, then you must be using an old MSVC compiler.  Your compiler appears to be Visual Studio 2013 or older.  I recommend using MSVC 2017.  It's free to download

badrik91, 2017-08-24:
Hello Sir,
I have faced problem with this two command scons -C CAMotics optimize=1 and scons -C CAMotics package which are given below .How to solve this?
I  have used visual studio  2017 and qt 5.8 .
Please give solution .
[image]

badrik91, 2017-08-24:
[image]
this one is second one.

jcoffland, 2017-08-25:
The first problem is caused by not have ``CBANG_HOME`` set.  The second is caused by not having ``QTDIR`` set.  Or perhaps these environment variables are set but they are set to the wrong location.  See https://github.com/CauldronDevelopmentLLC/CAMotics/blob/master/doc/Windows_Build.md

badrik91, 2017-08-26:
Thank you sir for giving reply.
but i have got this kind of error while run this command scons -C CAMotics package,
[image]
how to solve this?
Please give solution.

jcoffland, 2017-08-26:
If you followed the Windows build instructions then you should have installed NSIS and created a file called ``env.bat`` which added ``c:\Program Files\NSIS\`` to the ``PATH``.  Before running ``scons -C CAMotics package`` you should have run ``env.bat`` in the same console.  Then the build system should be able to find ``makensis`` at ``c:\Program Files\NSIS\makensis.exe``.  If you installed NSIS somewhere else then you will have to adjust your setup accordingly.

It looks like you might be close though.  Can you just run ``CAMotics.exe`` from ``c:\build\CAMotics\CAMotics.exe``?

Why are you building CAMotics from source instead of using the pre built packages at http://camotics.org/download.html?

badrik91, 2017-08-28:
Thank you Sir,i have completed successfully build  this project in windows .

jcoffland, 2017-08-28:
Excellent!

closed 2017-08-28 by jcoffland