CauldronDevelopmentLLC/CAMotics #248: ChakraCore builds under Windows 10 but dynamic linking fails

jcoffland, 2017-08-30:
Oddly CAMotics builds and runs fine with ChakraCore in Linux but on Windows the run-time fails.  There are errors when the ChakraCore API is accessed.  Presumably, it's unable to load the DLL and access the API functions.  I'm not an expert at building software for Windows by any means so we have abandoned ChakraCore support in favor of libv8 for now.

xiaozhaolook, 2017-12-01:
that's really a pity, I am considering to learn some technology and skill for simulate 3D print path solution, can you give me some advices? thanks in advance!

jcoffland, 2017-12-18:
@xiaozhaolook  you don't need ChakraCore to run CAMotics in Windows.  It works with Google's v8 Javascript engine.

jcoffland, 2019-01-19:
Not using ChakraCore any more.

closed 2019-01-19 by jcoffland