CauldronDevelopmentLLC/CAMotics #249: how to add function in  ui  of CAMotics and compile

badrik91, 2017-09-05:
I have a doubt ,how to add function and implement it by using Qt creator because you later told that we could not build in Qt.so this happens only on scons command tool ? or build on Qt also.

jcoffland, 2017-09-05:
It depends on what OS you are using.  Linux is the easiest for development.  It's also possible on macOS and Windows.  Here are instructions for building on Windows: https://github.com/CauldronDevelopmentLLC/CAMotics/blob/master/doc/Windows_Build.md and for Linux: https://github.com/CauldronDevelopmentLLC/CAMotics#building-from-source

badrik91, 2017-09-08:
Hello Sir,
I build this project using scon and its build sucessfully but when i add the event in menubar of ui part but not build in qt so i compiled using scon works fine but how to debug this code so i test this event? 
Please give solution about this .

jcoffland, 2017-09-08:
I don't know.  I only use Qt Creator for UI design.  I connect my events in C++ code.  Or with ``moc`` which scons runs for me.

Maybe if you tell me what you are trying to do I can help.

badrik91, 2017-09-08:
I want to insert one menu item like print in file menubar .whenever i clicked on print menu that call the exe .thats the exactly i want to do.Are you getting my point?
Please give soluition if possible.

jcoffland, 2017-09-08:
You should look at examples in the code.  There are lots of places where menu item events are used.  For example, https://github.com/CauldronDevelopmentLLC/CAMotics/blob/master/src/camotics/qt/QtWin.cpp#L1938

badrik91, 2017-09-08:
I already add menu in this file and its header.but i don't debug this menu because not build this in qt .so I compile this through scon and everything comes right but when i clicked the menu that does not work properly so i want to check this event that resolves when i debug the code.Is there any tool to debug the UI backed code?

jcoffland, 2017-09-08:
``printf()`` or gdb.

badrik91, 2017-09-08:
Printf do print the message.gdb not use for me because i want to do in windows.Is it possible to build in Qt  for debugging ? or Any another way to check ui event?

jcoffland, 2017-09-08:
I don't know.  I develop in Linux.

badrik91, 2017-09-09:
In Linux, Are you used Qt for UI design?and build sucessfully on QT ? or used scons for that.
I have sucessfully build this in linux using scon but not in QT.How to configure and build project in qt ?
Please give solution if possible

jcoffland, 2017-09-11:
I use Qt for UI design only.  Then I build with scons.

badrik91, 2017-09-11:
Are you sucessfully build in QT?If i sucessfully build in QT then  I can modify and check event .can you give solution about to build in qt?

jcoffland, 2017-09-11:
No.

closed 2023-08-10 by jcoffland