CauldronDevelopmentLLC/CAMotics #250: wrong display of the plunges

jkmnt, 2017-09-06:
The plunges (maybe with X/Y movement ?) are displayed wrong in 1.1.+ versions. It is very evident on the holding tabs, please see below:

v1.1.1:
[image]

v1.0.5:
[image]

used nc code & xml project:
[wrong_plunge_nc.zip](https://github.com/CauldronDevelopmentLLC/CAMotics/files/1283024/wrong_plunge_nc.zip)

jcoffland, 2017-09-07:
Thanks for the nice clean example.  I wish everyone did it like this. You are absolutely right.  I already found and fixed this bug.  The next release, which I hope to get out soon, fixes this problem.

I'm currently tied up with our Kickstarter campaign.  http://buildbotics.com/kickstarter

jkmnt, 2017-09-11:
Ok, thanks. Good luck with the campaign !

jcoffland, 2019-02-06:
Fixed in upcoming v1.2.0 release.

closed 2019-02-06 by jcoffland