CauldronDevelopmentLLC/CAMotics #251: OS X 10.12 build fatal error: 'type_traits' file not found

zrichmond, 2017-09-09:
```
...
In file included from /usr/local/Cellar/qt/5.9.1/include/QtGui/qtguiglobal.h:43:
/usr/local/Cellar/qt/5.9.1/include/QtCore/qglobal.h:45:12: fatal error: 'type_traits' file not found
...
```

As you might be able to tell from the above error I am using homebrew.
xcode CLT version 8.3.2

Digging deeper on this before it looked like it is related to xcode using an old version of gcc and the following might help:

[how-can-i-get-scons-to-use-clang-with-libc](https://stackoverflow.com/questions/25386551/how-can-i-get-scons-to-use-clang-with-libc)

One challenge I couldn't over come is my unfamiliarity with the scons build file SConstruct and I couldn't the right build flags to work without creating other errors.

jcoffland, 2017-09-11:
I ran into the same problem recently.  I'm not sure what option fixed it but my automated builds build both cbang and CAMotics like this:

```
scons compiler=clang cxxflags=-stdlib=libc++ linkflags=-stdlib=libc++ cxxstd=c++11 disable_local="libevent re2" osx_min_ver=10.7
```
Make sure you've got the latest cbang.

zrichmond, 2017-09-11:
That solved it!  On to a different issue now. 

Don't know if it's worth it yet, but I may dig into the SConstruct build file to see if I can get it to match without breaking something else and submit a pull request.

closed 2017-09-11 by zrichmond

jcoffland, 2017-09-11:
Awesome.  I definitely need some outside expertise here.