CauldronDevelopmentLLC/CAMotics #252: Windows 32bit version not startable

lome64, 2017-09-27:
After installing the 1.1.1 Windows 32bit version, "camotics.exe is not a valid 32bit application" pops up.

jcoffland, 2017-09-27:
What version of Windows are you using?  I just tried CAMotics 1.1.1 on 32-bit Windows 10 and it worked correctly.

lome64, 2017-09-28:
I must admit it's an older machine with the last version of Windows XP on it.
We keep it running in order to use attached hardware which has no drivers for more recent versions of Windows.
The version 1.0.6 (1.0.4?) however of Camotics worked fine on that machine.
Stictly it's not necessary to run Camotics on that old machine, but I noticed the problem and wanted to let you know.
Lothar

jcoffland, 2017-09-28:
Thanks for letting me know.  CAMotics 1.1.1 does not work on XP.  The next release will make this clear at install time.

closed 2017-09-28 by jcoffland