CauldronDevelopmentLLC/CAMotics #253: Nothing displays in Simulation view when running in Ubuntu 17.10

lslewis901, 2017-11-03:
Hi,

I downloaded the deb installer and then installed it.  It failed to run and I found a missing library error which I corrected by running the apt-get commands used for installing the source.  Now the desktop GUI loads, but none of the samples resent anything in the simulation view.  It remains black.

Thanks

jcoffland, 2017-11-03:
Are you running in a VM?  Are there any errors in the console.  Click ``View`` -> ``Show Console``.

lslewis901, 2017-11-03:
Hi Joseph,

This is a Thinkpad 500 (Intel video chip set) running Ubuntu.    If I run
it from a Term session this is what it shows...
No errors, except the missing tool message.

[image: Inline image 1]

[image: Inline image 3]

jcoffland, 2017-11-04:
The images you attached didn't come through but I believe you.  In the future, go to GitHub to reply and add images.  The problem may be your graphics driver.  You could try updating it.  You could also try going to ``Edit`` -> ``Settings`` -> ``Advanced`` and unchecking ``Enable surface VBOs`` and ``Enable path VBOs``.