CauldronDevelopmentLLC/CAMotics #254: Failed to load OpenGL

jorgemarmo, 2017-11-20:
Hi, I got this error when opening CAMotics 1.1.1,
```
Opening C:/Program Files (x86)/CAMotics/examples/camotics/camotics.xml
ERROR:Failed to load OpenGL.  You may need to upgrade your graphics driver.
ERROR:Initializing GLEW: Missing GL version
```

Architecture: core i5 2520M / intel HD 3000.
same problem in Windows 7 and Windows 10
latest intel drivers. All in x64

Other references to this problem:
[https://github.com/CauldronDevelopmentLLC/CAMotics/issues/159](https://github.com/CauldronDevelopmentLLC/CAMotics/issues/159)

[http://www.cambam.co.uk/forum/index.php?PHPSESSID=2d98be7e85696e533008fb23e6b29de0&topic=6881.0](http://www.cambam.co.uk/forum/index.php?PHPSESSID=2d98be7e85696e533008fb23e6b29de0&topic=6881.0)

[https://groups.google.com/forum/#!topic/camotics-users/Yjo3Zujx5tE](https://groups.google.com/forum/#!topic/camotics-users/Yjo3Zujx5tE)

jcoffland, 2017-11-27:
I believe the next release of CAMotics will fix this issue.  In the meantime, you might try upgrading your graphics driver.

jorgemarmo, 2017-12-03:
Hi, I see "OpenGL problems solved" on the v1.1.2 [changelog](https://github.com/CauldronDevelopmentLLC/CAMotics/blob/master/CHANGELOG.md), but I cannot find the v1.1.2 compiled, does it exist somewhere or is not available yet?

jcoffland, 2017-12-18:
@jorgemarmo The v1.1.2 release is not out yet.  I'm still working on it.

jorgemarmo, 2017-12-19:
@jcoffland  thank you, I'll look forward to test it.  ;)

jorgemarmo, 2017-12-19:
I'll close this issue and if I still have it on 1.1.2 I'll open a new one.

closed 2017-12-19 by jorgemarmo

jcoffland, 2019-06-18:
Newer versions CAMotics have problems with Intel 3000 and 4000 HD graphics drivers.  You can try an older version of CAMotics.  See #290.