CauldronDevelopmentLLC/CAMotics #258: rendering of cut made by cylindrical tool can be incorrect

foxharp, 2018-02-25:
i have a tool path which leaves a "tab" at the bottom of the cut, so that the two pieces which have been mostly separated are still connected by a small bit of material.  this "tab" is rendered incorrectly.  i'm running 1.1.1, installed from ubuntu 16.04.

first, the sample code, which simply descends the cutter, runs it across, and bumps it up in the middle to form the tab:

`
    ( tabtest )
    g17 (xy plane for circles)(g17:xy; g18:xz; g19 yz)
    g20 (inch units)(g20:inch; g21:mm)
    g40 (cancel cutter radius compensation)
    g49 (cancel tool length offset)
    g54 (coordinate system 1)(g54,g55,g56,g57,g58,g59,g59.1,g59.2,g59.3 are sys>
    g80 (cancel motion mode)
    g90 (absolute distance mode)
    g90.1 (absolute arc distance mode)
    g92.1 (reset g92 offsets)
    g94 (feed/minute mode)
    g97 (spindle speed is rpm, vs constant surface speed)
    m9 (all coolant off)
    m5 (spindle off) (program should turn on with m3 and s<spindle speed>)
    G0 F 80
    G1 F 13
    G0 Z .1
    G0 X 0 Y 0
    ( bite was .4, surface is .01, depth is -.5 )
    ( fixed bite is .256000 )
    ( passes is 2 )
    G0 Z .1
    ( outline to depth -.5 )
    ( top of loop, reverse is )
    G0 Z .1
    ( outline: forwarding )
    G0 X -1 Y 0
    G1 Z -.246000
    G1 X -.5 Y 0
    G1 X .5 Y 0
    G1 X 1 Y 0
    ( top of loop, reverse is )
    G0 Z .1
    ( outline: forwarding )
    G0 X -1 Y 0
    G1 Z -.5
    G1 X -.5 Y 0
    G91
    G0 Z .06
    G90
    G1 X .5 Y 0
    G91
    G1 Z -.06
    G90
    G1 X 1 Y 0
    G0 Z .1
    G40 (cancel cutter radius compensation)
    G92.1  (clear G92 offsets)
    M30 (End of Program)

`

and then, a screen shot.  i reduced the automatic workpiece margins to 0% to give a clear view.  clearly, both ends of the tab should have a concave shape, since they were formed by a round cutter.   note that the code above uses relative (G91) moves to "hop" over the tab.  the simulation gives the same result if absolute moves are used instead.

[image]

(thanks for camotics, btw -- it's a nice addition to the open source cad/cam tool suite!  and clearly was a _lot_ of work!)

jcoffland, 2018-03-01:
This is a duplicate of #250.  It has been fixed in the next version which I will hopefully be able to release soon.  If you don't already, please look for duplicates before submitting new issues.

closed 2018-03-01 by jcoffland

foxharp, 2018-03-01:
i did look, but only in open bugs.