CauldronDevelopmentLLC/CAMotics #262: Question about machine files

hpmachining, 2018-06-04:
Is there any documentation concerning the machine files? I assume they are configurations for specific CNC machines. Can you point me to any information on using / writing these files?

Thank you.

jcoffland, 2018-06-04:
This is still in development so no docs yet.  Your best resource is the code itself.  The machine descriptions are in JSON.  The format is a TCO file but I will probably convert it all to JSON before I'm done.  What is your end goal?

hpmachining, 2018-06-04:
Right now I was just wondering what they are for. I thought they might be a way to configure some machine specific g-codes or something like that.

jcoffland, 2018-06-04:
It will eventually allow for machine specific GCode.  The main purpose is to be able to define the geometry and movement parameters of the machine.  It's also possible to create a 3D model of the machine that can be visualized as part of the simulation.  It's a work in progress.

closed 2018-06-04 by jcoffland