CauldronDevelopmentLLC/CAMotics #263: symbol lookup error on Linux Mint 17.3

KiLLAAA, 2018-06-08:
Hi, i have installed camotics from current 64bit deb package on Linux Mint 17.3 and when i try to run it there goes error, then error and finally: error.
First i had to solve: [Missing libv8-3.14.5 on debian 8 64 #215](https://github.com/CauldronDevelopmentLLC/CAMotics/issues/215)

then: [libstdc++.so.6: version `GLIBCXX_3.4.20' not found #178](https://github.com/CauldronDevelopmentLLC/CAMotics/issues/178)

and now im facing:
`$ camotics
*********************************** CAMotics ***********************************
     Website: http://camotics.org/
   Copyright: (c) 2011-2015
      Author: Joseph Coffland <joseph@camotics.org>
Organization: Cauldron Development LLC
        Args: 
      Config: <none>
************************************ Build *************************************
     Version: 1.1.1
        Date: Feb 26 2017
        Time: 07:03:12
  Repository: Git
    Revision: 0349ce6eb88d1e63c77bc7a50dd5e0653036a16d
      Branch: master
    Compiler: GNU 4.9.2
     Options: -std=c++11 -Wno-deprecated-declarations -O3 -funroll-loops -fPIC
    Platform: linux2 4.9.0-1-amd64
        Bits: 64
        Mode: Release
************************************ System ************************************
         CPU: AMD Phenom(tm) II X4 955 Processor
      CPU ID: AuthenticAMD Family 16 Model 4 Stepping 2
        CPUs: 4
      Memory: 7.79GiB
 Free Memory: 1.88GiB
     Threads: POSIX_THREADS
  OS Version: 4.2
 Has Battery: false
  On Battery: false
  UTC Offset: 2
         PID: 29499
         CWD: /home/k
         DPI: 96
********************************************************************************
camotics: symbol lookup error: camotics: undefined symbol: _ZN7QString13toUtf8_helperERKS_`

**reinstalling package didnt help.. any hints to fix this symbol lookup error?**

jcoffland, 2018-06-08:
You may have to compile CAMotics from source to get it to work on that platform.  See https://stackoverflow.com/questions/27714446/undefined-symbol-zn7qstring13toutf8-helpererks-at-runtime

Capnden, 2018-10-17:
I had the exact same experience.  I only solved it by using an earlier release of Camotics available on their download site.  I believe it was release 1.0.8

jcoffland, 2019-02-09:
Fixed in the new v1.2.0 release.

closed 2019-02-09 by jcoffland