CauldronDevelopmentLLC/CAMotics #264: Rendering issues with real-time simulation.

hpmachining, 2018-06-10:
Rendering looks correct when using "Run", but incorrect when using "Play" button. It is worse the faster you run the simulation. Also, the renderings don't repeat when simulated multiple times (using "Play"). I get similar quality, but deviations differ.

**Run**
[image]

**Play 1x**
[image]

**Play 1x - 2nd example**
[image]

**Play 4x**
[image]

**System Information:**
Memory: 15.5GiB
Processor: Intel® Core™ i5-8250U CPU @ 1.60GHz × 8
Graphics: Intel® UHD Graphics 620 (Kabylake GT2) 
OS - Arch Linux 64bit with GNOME 3.28.2
CAMotics 1.2.0 commit 5c02a34
V8 3.14.5
Qt 5.11.0

jcoffland, 2018-06-10:
Yes, I was aware of this.  Thanks for opening an issue for it though.  When "Play" is active CAMotics attempts to simulate as the tool moves but the simulation cannot keep up unless the simulation is very simple or the movement slowed way down.  So it attempts to simulate only the area of the workpiece that has changed since the last simulation was run.  Unfortunately, it doesn't always run these localized simulations correctly.  I believe the problem is due to incorrectly calculating the area that needs to be restimulated.

hpmachining, 2018-06-11:
Thank you for the information. I thought it had something to do with not keeping up. I understand that it is a work in progress. Thank you for all you have done with this project. I have been studying the code and hope to learn how you do the simulations, but for now I'm just playing around with the UI. See my [fork](https://github.com/hpmachining/CAMotics) if you are interested in anything I have done.

jcoffland, 2018-06-11:
Awesome, I merged all of your changes.  You may want to resync your fork.

jcoffland, 2019-02-09:
Fixed in new v1.2.0 release.

closed 2019-02-09 by jcoffland