CauldronDevelopmentLLC/CAMotics #270: Can CAMotics run on Python 3?

srakitnican, 2018-08-26:
Fedora is slowly getting rid of Python 2. It seems CAMotics is using Python only for the scons build scripts. Could the scripts work with Python 3?

jcoffland, 2018-08-26:
I haven't actually tried it yet but I just ran pylint3 on the relavant parts and made some corrections to the build scripts.  I'd be happy to help if there is anything else needed to support python3.

srakitnican, 2019-03-28:
Hi, sorry that I could not test this earlier. I able to run it now using SCons that runs on Python 3.

Issues found:
```
Checking for C header file bfd.h... no
ImportError: cannot import name 'maketrans' from 'string' (/usr/lib64/python3.7/string.py):
  File "/builddir/build/BUILD/cbang-1.3.2/SConstruct", line 65:
    variant_dir = 'build/' + lib))
  File "/usr/lib/python3.7/site-packages/SCons/Script/SConscript.py", line 667:
    return method(*args, **kw)
  File "/usr/lib/python3.7/site-packages/SCons/Script/SConscript.py", line 604:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/usr/lib/python3.7/site-packages/SCons/Script/SConscript.py", line 285:
    call_stack[-1].globals)
  File "/builddir/build/BUILD/cbang-1.3.2/build/libevent/SConscript", line 318:
    defs = get_event_config_defs()
  File "/builddir/build/BUILD/cbang-1.3.2/build/libevent/SConscript", line 193:
    from string import maketrans
```
maketrans is in standard library in Python 3, also https://stackoverflow.com/a/27540139
Modified it to get past
```
    #from string import maketrans
    trans = "".maketrans('/. *', '___P')
```
The next issue:
```
AttributeError: 'filter' object has no attribute 'name':
  File "/builddir/build/BUILD/cbang-1.3.2/SConstruct", line 65:
    variant_dir = 'build/' + lib))
  File "/usr/lib/python3.7/site-packages/SCons/Script/SConscript.py", line 667:
    return method(*args, **kw)
  File "/usr/lib/python3.7/site-packages/SCons/Script/SConscript.py", line 604:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/usr/lib/python3.7/site-packages/SCons/Script/SConscript.py", line 285:
    call_stack[-1].globals)
  File "/builddir/build/BUILD/cbang-1.3.2/build/libevent/SConscript", line 355:
    lib = env.Library('#/lib/event', src)
  File "/usr/lib/python3.7/site-packages/SCons/Environment.py", line 260:
    return MethodWrapper.__call__(self, target, source, *args, **kw)
  File "/usr/lib/python3.7/site-packages/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/usr/lib/python3.7/site-packages/SCons/Builder.py", line 645:
    return self._execute(env, target, source, OverrideWarner(kw), ekw)
  File "/usr/lib/python3.7/site-packages/SCons/Builder.py", line 551:
    source = self.src_builder_sources(env, source, overwarn)
  File "/usr/lib/python3.7/site-packages/SCons/Builder.py", line 751:
    match_suffix = match_src_suffix(s.name)
```

jcoffland, 2019-03-28:
These are cbang issues.  I believe I've fixed them but getting these fixes will require updating to the latest CAMotics code.  Let me know if there's anything else before I tag a new cbang version.

srakitnican, 2019-03-28:
```
Checking for C library kern... no
<map object at 0x7f81f65413c8>
AttributeError: 'map' object has no attribute 'append':
  File "/builddir/build/BUILD/cbang-1.3.2/SConstruct", line 97:
    src.append(res)
```

jcoffland, 2019-03-28:
Fixed.

srakitnican, 2019-03-28:
It starts compiling now, this pops up in the process:

```
g++ -o build/cbang/event/WebServer.o -c -std=c++11 -g -Wall -fno-pie -I/usr/include/v8-3.14/ -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -Wno-error=parentheses -Wno-deprecated-declarations -fPIC -DDEBUG -D_REENTRANT -DHAVE_EXPAT -DHAVE_PTHREADS -DHAVE_LIBSQLITE -DHAVE_V8 -DHAVE_VALGRIND -DDEBUG_LEVEL=1 -DUSING_CBANG -Iinclude -Isrc -Isrc/boost src/cbang/event/WebServer.cpp
resources_build(["build/resources.cpp", "build/resources.data/data2.cpp", "build/resources.data/data3.cpp", "build/resources.data/data4.cpp", "build/resources.data/data5.cpp", "build/resources.data/data6.cpp", "build/resources.data/data7.cpp", "build/resources.data/data8.cpp", "build/resources.data/data9.cpp", "build/resources.data/data10.cpp"], ["src/resources"])
scons: *** [build/resources.cpp] TypeError : a bytes-like object is required, not 'str'
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/SCons/Action.py", line 1202, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "/builddir/build/BUILD/cbang-1.3.2/config/resources/__init__.py", line 189, in resources_build
    f = start_file(ctx, target)
  File "/builddir/build/BUILD/cbang-1.3.2/config/resources/__init__.py", line 29, in start_file
    '\\' + ('*' * 75) + '/\n'
TypeError: a bytes-like object is required, not 'str'
scons: building terminated because of errors.

```

jcoffland, 2019-03-28:
Should be fixed.  Unfortunately, I don't have an scons for Python 3 setup.

srakitnican, 2019-03-28:
```
Writing resource: src/resources/licenses/zlib.txt to build/resources.data/data2.cpp
scons: *** [build/resources.cpp] TypeError : ord() expected string of length 1, but int found
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/SCons/Action.py", line 1202, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "/builddir/build/BUILD/cbang-1.3.2/config/resources/__init__.py", line 192, in resources_build
    write_resource(ctx, f, data_dir, str(src))
  File "/builddir/build/BUILD/cbang-1.3.2/config/resources/__init__.py", line 84, in write_resource
    child_resources, exclude)
  File "/builddir/build/BUILD/cbang-1.3.2/config/resources/__init__.py", line 84, in write_resource
    child_resources, exclude)
  File "/builddir/build/BUILD/cbang-1.3.2/config/resources/__init__.py", line 111, in write_resource
    write_string(ctx, out, '%d,' % ord(c))
TypeError: ord() expected string of length 1, but int found
scons: building terminated because of errors.

```

jcoffland, 2019-03-28:
Hopefully I got it this time.

srakitnican, 2019-03-28:
Removing binary mode seems to also do the trick. That seems to be everything from cbang. Following camotics.

```
scons: Reading SConscript files ...
TypeError: cannot use a string pattern on a bytes-like object:
  File "/builddir/build/BUILD/CAMotics-1.2.0/SConstruct", line 37:
    env.Tool('qt5', toolpath = ['./config'])
  File "/usr/lib/python3.7/site-packages/SCons/Environment.py", line 1789:
    tool(self)
  File "/usr/lib/python3.7/site-packages/SCons/Tool/__init__.py", line 296:
    self.generate(env, *args, **kw)
  File "/builddir/build/BUILD/CAMotics-1.2.0/config/qt5/__init__.py", line 701:
    env['QT5DIR'] = _detect(env)
  File "/builddir/build/BUILD/CAMotics-1.2.0/config/qt5/__init__.py", line 426:
    vernumber = mocver_re.match(vernumber)
error: Bad exit status from /var/tmp/rpm-tmp.VQC0mL (%build)
    Bad exit status from /var/tmp/rpm-tmp.VQC0mL (%build)
```

srakitnican, 2019-03-29:
Seems like the SCons Qt5 tool has been ported upstream already:
https://bitbucket.org/dirkbaechle/scons_qt5/issues/10/python3-support

I'll see if this works

jcoffland, 2019-03-29:
I got pulled away to something else yesterday.  I believe I've fixed the above.

srakitnican, 2019-03-30:
After applying the fix this is the next issue:
```
DXFLib missing required features(cached) error: no result
Checking for C library stdc++... yes
AttributeError: 'map' object has no attribute 'name':
  File "/builddir/build/BUILD/CAMotics-1.2.0.scons3fixes/SConstruct", line 133:
    lib = env.Library('build/libGCode', src)
  File "/usr/lib/python3.7/site-packages/SCons/Environment.py", line 260:
    return MethodWrapper.__call__(self, target, source, *args, **kw)
  File "/usr/lib/python3.7/site-packages/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/usr/lib/python3.7/site-packages/SCons/Builder.py", line 645:
    return self._execute(env, target, source, OverrideWarner(kw), ekw)
  File "/usr/lib/python3.7/site-packages/SCons/Builder.py", line 551:
    source = self.src_builder_sources(env, source, overwarn)
  File "/usr/lib/python3.7/site-packages/SCons/Builder.py", line 751:
    match_suffix = match_src_suffix(s.name)
```

jcoffland, 2019-03-30:
Ok, I fixed that and a few other similar problems.

srakitnican, 2019-03-30:
```
scons: done reading SConscript files.
scons: Building targets ...
g++ -o build/camotics.o -c -std=c++11 -Wno-deprecated-declarations -g -Wall -fno-pie -I/usr/include/v8-3.14/ -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -Wno-error=parentheses -fPIC -DDEBUG -D_REENTRANT -DHAVE_EXPAT -DHAVE_PTHREADS -DHAVE_LIBSQLITE -DHAVE_V8 -DHAVE_VALGRIND -DDEBUG_LEVEL=1 -DHAVE_CBANG -DUSING_CBANG -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_NETWORK_LIB -DQT_WIDGETS_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DQT_COMPILING_QSTRING_COMPAT_CPP -DCAMOTICS_GUI -I/builddir/build/BUILD/cbang-1.3.2.scons3fixes/src -I/builddir/build/BUILD/cbang-1.3.2.scons3fixes/include -I/builddir/build/BUILD/cbang-1.3.2.scons3fixes/src/boost -Isrc -Ibuild -Isrc src/camotics.cpp
scons: *** [build/qrc_camotics.o] TypeError : cannot use a string pattern on a bytes-like object
scons: building terminated because of errors.

```

jcoffland, 2019-04-02:
Ok, I finally set up my own scons with Python 3 build environment.  I made several more changes to both cbang and CAMotics but it should be working now.

srakitnican, 2019-04-03:
Works now, thanks!