CauldronDevelopmentLLC/CAMotics #273: Flatpak?

rendeko, 2018-10-01:
Since building from source can be finicky (V8/ChakraCore needing to be exported variables for cbang to compile properly, not much documentation about V8/ChakraCore in general) as well as the Debian binaries being built for an older version of libssl, has there been any consideration into building a flatpak?

This would essentially create a distro-universal binary with no worries about users sourcing certain versions of dependencies. I think this would be ideal for the work scenarios that CAMotics is useful for, where CAMotics can stay on stable binaries in it's own sandbox while the rest of the system can benfit from cutting edge upgrades.

jcoffland, 2019-01-19:
http://docs.flatpak.org/

luzpaz, 2022-06-27:
This ticket should have a 'packaging' label

luzpaz, 2022-07-10:
CC @hfiguiere (would you know of anyone who would have any interest creating and maintaining a CAMotics flatpak?)

hfiguiere, 2022-07-11:
Apparently there was some work done with #347. Not sure why it never got submitted.

tpimh, 2022-07-11:
I have published my work here: [camotics-flatpak](https://github.com/tpimh/camotics-flatpak). I have never finished it, but feel free to test, and any contributions are welcome!

luzpaz, 2022-07-11:
@tpimh any specific reason why you haven't finished? Is it just personal availability or (a) technical issue(s)?

tpimh, 2022-07-12:
To be honest, I'm not sure what exactly is not working right now. I think there were some technical issues, but I have successfully run some examples. It's best to build it (instructions included in the repo) and test.