CauldronDevelopmentLLC/CAMotics #276: only tool table #1?

alexvincon, 2018-11-07:
Hi how can I import some gcode and change the tool. Every time I import any g code it dont recognize the tool.
Thanks

jcoffland, 2018-11-07:
What you create a new project it asks you which tool table to use.  The options are "Default", "Current", "Empty".  You can create a "Default" tool table by setting up the tool table how you want it then click "Tools" -> "Save Tool Table As Default".

alexvincon, 2018-11-08:
thanks but thats is whst i make and every time i lo ad a project it dont recohnize the tool

jcoffland, 2018-11-08:
I just tested this and it seems to work.  At least in my version of the software which is a bit newer than what is currently online.  Do the tool numbers in your GCode match tool numbers in your default tool table?  What errors are you seeing?  Please post them here.