CauldronDevelopmentLLC/CAMotics #278: Installing on Raspbian (Raspberry Pi)

ChristianF88, 2018-12-29:
Hi, 
I'm trying to install the software on a raspberry pi running Raspbian.
I tried install the 32 bit Linux version.
After adding the i386 architecture I now get the following error.
```
sudo gdebi camotics_1.1.1_i386.deb

Reading package lists... Done
Building dependency tree        
Reading state information... Done
Reading state information... Done
Dieses Paket kann nicht installiert werden
Dependency is not satisfiable: libc6
```

I also tried building from source and I got the error:

```
# Install the prerequisites
sudo apt-get update
sudo apt-get -y install scons build-essential libssl-dev qt5-default \
  libqt5websockets5-dev libqt5opengl5-dev libv8-dev git

# Get the source
git clone https://github.com/CauldronDevelopmentLLC/cbang
git clone https://github.com/CauldronDevelopmentLLC/CAMotics

# Build C!
scons -C cbang
export CBANG_HOME=$PWD/cbang

# Build CAMotics
cd CAMotics
scons

g++ -o build/camotics/machine/MachineModel.o -c -std=c++11 -Wno-deprecated-declarations -O3 -funroll-loops -fPIC -DNDEBUG -D_REENTRANT -DHAVE_EXPAT -DHAVE_PTHREADS -DHAVE_LIBSQLITE -DHAVE_OPENSSL -DHAVE_V8 -DHAVE_CBANG -DUSING_CBANG -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_NETWORK_LIB -DQT_WIDGETS_LIB -DQT_NO_OPENGL_ES_2 -DGL_GLEXT_PROTOTYPES -DCAMOTICS_GUI -I/home/pi/Downloads/cbang/src -I/home/pi/Downloads/cbang/include -I/home/pi/Downloads/cbang/src/boost -Isrc -I/usr/include/arm-linux-gnueabihf/qt5/QtOpenGL -I/usr/include/arm-linux-gnueabihf/qt5 -I/usr/include/arm-linux-gnueabihf/qt5/QtWidgets -I/usr/include/arm-linux-gnueabihf/qt5/QtGui -I/usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -I/usr/include/arm-linux-gnueabihf/qt5/QtNetwork -I/usr/include/arm-linux-gnueabihf/qt5/QtCore -Ibuild -Isrc src/camotics/machine/MachineModel.cpp
In file included from /usr/include/arm-linux-gnueabihf/qt5/QtGui/qopenglversionfunctions.h:62:0,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtGui/QOpenGLVersionFunctions:1,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtGui/qopenglfunctions_2_1.h:57,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtGui/QOpenGLFunctions_2_1:1,
                 from src/camotics/view/GL.h:23,
                 from src/camotics/contour/TriangleSurface.cpp:31:
/usr/include/arm-linux-gnueabihf/qt5/QtGui/qopengl.h:129:21: fatal error: GL/gl.h: Datei oder Verzeichnis nicht gefunden
 #  include <GL/gl.h>
                     ^
compilation terminated.
scons: *** [build/camotics/contour/TriangleSurface.o] Error 1
In file included from /usr/include/arm-linux-gnueabihf/qt5/QtGui/qopenglversionfunctions.h:62:0,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtGui/QOpenGLVersionFunctions:1,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtGui/qopenglfunctions_2_1.h:57,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtGui/QOpenGLFunctions_2_1:1,
                 from src/camotics/view/GL.h:23,
                 from src/camotics/machine/MachineModel.cpp:24:
/usr/include/arm-linux-gnueabihf/qt5/QtGui/qopengl.h:129:21: fatal error: GL/gl.h: Datei oder Verzeichnis nicht gefunden
 #  include <GL/gl.h>
                     ^
compilation terminated.
scons: *** [build/camotics/machine/MachineModel.o] Error 1
scons: building terminated because of errors.
```
Do you have any suggestions? :)

Thanks a bunch!

jcoffland, 2018-12-29:
Looks like you're missing some OpenGL development libraries.  Try installing ``libgl-dev``.

CAMotics will not work well on the RPi, even if you do get it to compile there.  The RPi does not have enough power to run the simulations effectively.

Running the ``i386`` arch on the RPi, if it even works, would be extremely slow because it would have to be emulated.  The RPi has an ARM processor which is incompatible with i386.

ChristianF88, 2018-12-29:
Thanks! Now I got a bit further! But still not all the way...
 ```
git clone https://github.com/CauldronDevelopmentLLC/cbang.git
scons -C cbang
export CBANG_HOME=$PWD/cbang

git clone https://github.com/CauldronDevelopmentLLC/CAMotics.git
cd CAMotics
scons

scons package
```

```
scons: warning: QT5DIR variable is not defined, using moc executable as a hint (QT5DIR=/usr)
File "/home/pi/CAMotics/config/qt5/__init__.py", line 439, indetect

scons: Reading SConscript files ...
  Compiler: gcc (gnu)
  Platform: posix
      Mode: gnu
      Arch: x86
      Jobs: 4
Checking for C library pthread... (cached) yes
Checking for C header file zlib.h... (cached) yes
Checking for C library z... (cached) yes
Checking for C header file bzlib.h... (cached) yes
Checking for C library bz2... (cached) yes
Checking for C header file expat.h... (cached) yes
Checking for C library expat... (cached) yes
Checking for C header file pthread.h... (cached) yes
Checking for C library pthread... (cached) yes
Checking for C library dl... (cached) yes
Checking for C library sqlite3... (cached) yes
Checking for C header file sqlite3.h... (cached) yes
Checking for C function sqlite3_backup_init()... (cached) yes
Checking for C header file event.h... (cached) yes
Checking for C library event... (cached) yes
Checking for C++ header file re2/re2.h... (cached) yes
Checking for C library re2... (cached) yes
Checking for C header file yaml.h... (cached) yes
Checking for C library yaml... (cached) yes
Checking for C library leveldb... no
Checking for C header file mysql/mysql.h... no
Checking for C function clock_gettime()... (cached) yes
Checking for C library dl... (cached) yes
Checking for C header file openssl/ssl.h... (cached) yes
Checking for C library crypto... (cached) yes
Checking for C library ssl... (cached) yes
Checking for openssl version >= 1.0.0...(cached) yes
Checking for C++ header file ChakraCore.h... no
Need C++ header ChakraCore.h(cached) error: no result
Checking for C++ header file v8.h... (cached) yes
Checking for C library v8... (cached) yes
...
installing "examples/wave/wave.camotics" to "build/camotics-deb/usr/share/doc/camotics/examp$
installing "examples/wave/wave.gcode" to "build/camotics-deb/usr/share/doc/camotics/examples$
installing "examples/wave/wave.tpl" to "build/camotics-deb/usr/share/doc/camotics/examples/w$
installing "machines/LICENSE.md" to "build/camotics-deb/usr/share/doc/camotics/machines/LICE$
installing "machines/bf20.json" to "build/camotics-deb/usr/share/doc/camotics/machines/bf20.$
installing "machines/bf20.tco" to "build/camotics-deb/usr/share/doc/camotics/machines/bf20.t$
installing "machines/cobra.json" to "build/camotics-deb/usr/share/doc/camotics/machines/cobr$
installing "machines/cobra.tco" to "build/camotics-deb/usr/share/doc/camotics/machines/cobra$
installing "machines/dummy.json" to "build/camotics-deb/usr/share/doc/camotics/machines/dumm$
installing "machines/dummy.tco" to "build/camotics-deb/usr/share/doc/camotics/machines/dummy$
installing "machines/milly.json" to "build/camotics-deb/usr/share/doc/camotics/machines/mill$
installing "machines/milly.tco" to "build/camotics-deb/usr/share/doc/camotics/machines/milly$
installing "machines/miss_marple.json" to "build/camotics-deb/usr/share/doc/camotics/machine$
installing "machines/miss_marple.tco" to "build/camotics-deb/usr/share/doc/camotics/machines$
installing "machines/shapeoko_2.json" to "build/camotics-deb/usr/share/doc/camotics/machines$
installing "machines/shapeoko_2.tco" to "build/camotics-deb/usr/share/doc/camotics/machines/$
installing "machines/shapeoko_3.json" to "build/camotics-deb/usr/share/doc/camotics/machines$
installing "machines/shapeoko_3.tco" to "build/camotics-deb/usr/share/doc/camotics/machines/$
installing "machines/sprite.json" to "build/camotics-deb/usr/share/doc/camotics/machines/spr$
installing "machines/sprite.tco" to "build/camotics-deb/usr/share/doc/camotics/machines/spri$
installing "machines/taig.json" to "build/camotics-deb/usr/share/doc/camotics/machines/taig.$
installing "machines/taig.tco" to "build/camotics-deb/usr/share/doc/camotics/machines/taig.t$
installing "machines/x-carve.json" to "build/camotics-deb/usr/share/doc/camotics/machines/x-$
installing "machines/x-carve.tco" to "build/camotics-deb/usr/share/doc/camotics/machines/x-c$
installing "machines/yohudi.json" to "build/camotics-deb/usr/share/doc/camotics/machines/yoh$
installing "machines/yohudi.tco" to "build/camotics-deb/usr/share/doc/camotics/machines/yohu$
installing "gcodetool" to "build/camotics-deb/usr/bin/gcodetool"
installing "camsim" to "build/camotics-deb/usr/bin/camsim"
scons: *** [camotics_1.2.0_armhf.deb] camsim: No such file or directory
scons: building terminated because of errors.

```

Any further suggestions?

Thanks a bunch! :)

jcoffland, 2018-12-29:
The ``scons`` build did not complete.  You cannot run ``scons package`` until ``scons`` works.

ChristianF88, 2018-12-30:
Also True! 

```
scons strict=0 > mysconslog.txt
```
console output:
```
scons: warning: QT5DIR variable is not defined, using moc executable as a hint (QT5DIR=/usr)
File "/home/pi/Programs/CAMotics/config/qt5/__init__.py", line 439, in _detect
libCAMotics.a(GL.o): In function `CAMotics::getGLFuncs()':
GL.cpp:(.text+0x114): undefined reference to `QOpenGLFunctions_2_1::versionProfile()'
collect2: error: ld returned 1 exit status
scons: *** [camsim] Error 1
libCAMotics.a(GL.o): In function `CAMotics::getGLFuncs()':
GL.cpp:(.text+0x114): undefined reference to `QOpenGLFunctions_2_1::versionProfile()'
collect2: error: ld returned 1 exit status
scons: *** [camotics] Error 1
```
The rest is in the attatched file.

Thanks! 

[mysconslog.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/2717050/mysconslog.txt)

ChristianF88, 2018-12-30:
Also True! The error occurs even earlier, sorry I didnt notice before.

```
scons -C cbang strict=0 > mysconscbanglog.txt
```

console output:
```
src/boost/libs/filesystem/src/operations.cpp: In function ‘int {anonymous}::readdir_r_simulator(DIR*, dirent*, dirent**)’:
src/boost/libs/filesystem/src/operations.cpp:2089:18: warning: ‘int readdir_r(DIR*, dirent*, dirent**)’ is deprecated [-Wdeprecated-declarations]
       { return ::readdir_r(dirp, entry, result); }
                  ^~~~~~~~~
In file included from /usr/include/features.h:364:0,
                 from /usr/include/arm-linux-gnueabihf/c++/6/bits/os_defines.h:39,
                 from /usr/include/arm-linux-gnueabihf/c++/6/bits/c++config.h:507,
                 from /usr/include/c++/6/cstddef:49,
                 from src/boost/boost/config/compiler/gcc.hpp:165,
                 from src/boost/boost/config.hpp:39,
                 from src/boost/boost/filesystem/operations.hpp:18,
                 from src/boost/libs/filesystem/src/operations.cpp:44:
/usr/include/dirent.h:189:12: note: declared here
 extern int __REDIRECT (readdir_r,
            ^
src/boost/libs/filesystem/src/operations.cpp:2089:47: warning: ‘int readdir_r(DIR*, dirent*, dirent**)’ is deprecated [-Wdeprecated-declarations]
       { return ::readdir_r(dirp, entry, result); }
                                               ^
In file included from /usr/include/features.h:364:0,
                 from /usr/include/arm-linux-gnueabihf/c++/6/bits/os_defines.h:39,
                 from /usr/include/arm-linux-gnueabihf/c++/6/bits/c++config.h:507,
                 from /usr/include/c++/6/cstddef:49,
                 from src/boost/boost/config/compiler/gcc.hpp:165,
                 from src/boost/boost/config.hpp:39,
                 from src/boost/boost/filesystem/operations.hpp:18,
                 from src/boost/libs/filesystem/src/operations.cpp:44:
/usr/include/dirent.h:189:12: note: declared here
 extern int __REDIRECT (readdir_r,
            ^
In file included from /usr/include/c++/6/vector:69:0,
                 from src/cbang/debug/StackTrace.h:37,
                 from src/cbang/Exception.h:39,
                 from src/cbang/config/Constraint.h:35,
                 from src/cbang/config/Option.h:36,
                 from src/cbang/config/Option.cpp:33:
/usr/include/c++/6/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_insert_aux(std::vector<_Tp, _Alloc>::iterator, const _Tp&) [with _Tp = long long int; _Alloc = std::allocator<long long int>]’:
/usr/include/c++/6/bits/vector.tcc:326:5: note: parameter passing for argument of type ‘std::vector<long long int>::iterator {aka __gnu_cxx::__normal_iterator<long long int*, std::vector<long long int> >}’ will change in GCC 7.1
     vector<_Tp, _Alloc>::
     ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/6/vector:64:0,
                 from src/cbang/debug/StackTrace.h:37,
                 from src/cbang/Exception.h:39,
                 from src/cbang/config/Constraint.h:35,
                 from src/cbang/config/Option.h:36,
                 from src/cbang/config/Option.cpp:33:
/usr/include/c++/6/bits/stl_vector.h: In static member function ‘static cb::Option::integers_t cb::Option::parseIntegers(const string&, const string&)’:
/usr/include/c++/6/bits/stl_vector.h:926:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<long long int*, std::vector<long long int> >’ will change in GCC 7.1
    _M_insert_aux(end(), __x);
    ^~~~~~~~~~~~~
In file included from /usr/include/c++/6/vector:69:0,
                 from src/cbang/debug/StackTrace.h:37,
                 from src/cbang/Exception.h:39,
                 from src/cbang/config/Constraint.h:35,
                 from src/cbang/config/Option.h:36,
                 from src/cbang/config/Option.cpp:33:
/usr/include/c++/6/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_insert_aux(std::vector<_Tp, _Alloc>::iterator, const _Tp&) [with _Tp = double; _Alloc = std::allocator<double>]’:
/usr/include/c++/6/bits/vector.tcc:326:5: note: parameter passing for argument of type ‘std::vector<double>::iterator {aka __gnu_cxx::__normal_iterator<double*, std::vector<double> >}’ will change in GCC 7.1
     vector<_Tp, _Alloc>::
     ^~~~~~~~~~~~~~~~~~~
/usr/include/c++/6/bits/vector.tcc:326:5: note: parameter passing for argument of type ‘std::vector<double>::iterator {aka __gnu_cxx::__normal_iterator<double*, std::vector<double> >}’ will change in GCC 7.1
In file included from /usr/include/c++/6/vector:64:0,
                 from src/cbang/debug/StackTrace.h:37,
                 from src/cbang/Exception.h:39,
                 from src/cbang/config/Constraint.h:35,
                 from src/cbang/config/Option.h:36,
                 from src/cbang/config/Option.cpp:33:
/usr/include/c++/6/bits/stl_vector.h: In static member function ‘static cb::Option::doubles_t cb::Option::parseDoubles(const string&, const string&)’:
/usr/include/c++/6/bits/stl_vector.h:926:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double> >’ will change in GCC 7.1
    _M_insert_aux(end(), __x);
    ^~~~~~~~~~~~~

```

The rest is in the attatched file.

Thanks! 

[mysconscbanglog.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/2717060/mysconscbanglog.txt)

jcoffland, 2019-01-02:
cbang is compiling correctly.  CAMotics would build because your Qt library is incompatible.  It's missing ``QOpenGLFunctions_2_1::versionProfile()``.

ChristianF88, 2019-01-03:
Ok, so that means I need to cross compile Qt for the Raspberry pi, since I cannot find any suitable libraries... right? and then try again?

Thank you for your help!

koendv, 2019-10-02:
You can download camotics as a debian package for Raspberry 4 from [github](https://github.com/koendv/camotics-raspberrypi).  HTH.

closed 2023-08-10 by jcoffland