CauldronDevelopmentLLC/CAMotics #280: Crash

sk8nfool, 2019-01-14:
Camotics crashed when adding (first) gcode file to new project. To replicate

- start camotics
- create new project (file->new project)
- poke OK in dialog
- add file (file->Add File)
- select gcode
- poke OK

This causes a crash with the message
```
camotics: build/cairo/cairo-surface.c:406: cairo_surface_reference: Assertion `((*&(&surface-ref_count)->ref_count) > 0)` failed.
```
I'm using UBUNTU 18.04 and CAMOTICS 1.1.1 on and amd64
core file available in my dropbox ([here](https://www.dropbox.com/s/whfg6po4trdpwb3/core?dl=0)).

sk8nfool, 2019-01-14:
This makes camotics useless to me. I can't even get started.

jcoffland, 2019-01-14:
Scratch that.  I was able to reproduce the bug with the debug version.  This appears to be fixed in the latest development version.  I'll try to get a new release out soon.  Thanks for the report.

In the meantime, you could try adding the file to the XML manually.  It's ugly but it should get you going.  It's the open file dialog that causes the problem.

jcoffland, 2019-02-09:
Fixed in the new v1.2.0 release.

closed 2019-02-09 by jcoffland