CauldronDevelopmentLLC/CAMotics #281: Use JSON as a format for importing feeds and speeds for materials

WillAdams, 2019-01-16:
Discussed at: https://groups.google.com/forum/#!topic/camotics-users/qYYuiPXYKVI

Considerations include:

 - how to determine feeds and speeds and where to source numbers from
 - relationships of them for a given selection of endmills 
 - which machine(s) should be supported

Rather than material, should there just be a set of files by chipload and desired feed rate?

Perhaps the machine setting in Autodesk Fusion 360 could be used?

jcoffland, 2019-01-16:
I do plan on implementing a feed and speed calculator in CAMotics.   Is that what you had in mind with feeds and speeds import?

WillAdams, 2019-01-17:
A feed and speed calculator would be _awesome_ --- I'd just like some easy way to have a .tpl file be easily switchable to cutting with a different material which has different feeds and speeds (in particular, I'm doing test cuts in soft fir/pine now, but will be switching to various tropical hardwoods for the actual project).

Similarly, some projects will switch between my Shapeoko and Nomad --- it would be nice to have a way to alter the feeds and speeds accordingly.

WillAdams, 2019-01-17:
Alternately, perhaps it could be implemented dynamically with some consideration of the underlying physics?

When I first heard of G-Wizard my assumption was that it would accept a G-Code file and once informed of the material which it was for, would analyze each movement and determine along that path how much material was present and how much force was necessary and would adjust the feed rate along the path (breaking it up if need be) and determine the optimal speed for the entire file (for machines which couldn't dynamically adjust the spindle speed).

It would be great to denote a desired chipload and cutting force when setting a path and have the program determine the appropriate feed rate for a given spindle speed.

jcoffland, 2019-01-17:
That is ultimately one of my goals for CAMotics.  Since CAMotics simulates the tool path cuts it would be possible to also compute the contact area between the tool and workpiece at any given moment.  The direction vector and speed are already known.  With some information about the tool and material, you can calculate lots of interesting things such as the force required, chip load, tool wear, etc.

The most difficult part is to compute the tool to workpiece contact area efficiently.  I have some ideas on how to estimate this accurately.  I just need time to get around to it.

WillAdams, 2019-01-18:
Very cool!

Apparently Autodesk uses JSON for tool definitions as well: https://support.bantamtools.com/hc/en-us/articles/115001671594-Fusion-360-Tool-Library