CauldronDevelopmentLLC/CAMotics #283: CAMotics 1.1.1 doesn't work with new cbang

srakitnican, 2019-02-03:
CAMotics 1.1.1 does not compile with newer cbang commit https://github.com/CauldronDevelopmentLLC/cbang/commit/a4e236e8db124a02d73f8a5bca1e82ad44a40b6c . Reverting to https://github.com/CauldronDevelopmentLLC/cbang/commit/fdd8867333016a479b66dcb3ec9de5eaf54b9adb works. GCC 9 was used.
```
g++ -o build/camotics/gcode/Parser.o -c -std=c++11 -Wno-deprecated-declarations -g -Wall -I/usr/include/v8-3.14/ -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -Wno-error=parentheses -fPIC -DDEBUG -D_REENTRANT -DHAVE_EXPAT -DHAVE_PTHREADS -DHAVE_LIBSQLITE -DHAVE_V8 -DDEBUG_LEVEL=1 -DHAVE_CBANG -DUSING_CBANG -DQT_CORE_LIB -DQT_GUI_LIB -DQT_OPENGL_LIB -DQT_SHARED -I/builddir/build/BUILD/cbang-a4e236e8db124a02d73f8a5bca1e82ad44a40b6c/src -I/builddir/build/BUILD/cbang-a4e236e8db124a02d73f8a5bca1e82ad44a40b6c/include -I/builddir/build/BUILD/cbang-a4e236e8db124a02d73f8a5bca1e82ad44a40b6c/src/boost -I/usr/include/QtCore -I/usr/include/QtGui -I/usr/include/QtOpenGL -Isrc -Ibuild -Isrc src/camotics/gcode/Parser.cpp
In file included from src/camotics/gcode/Tokenizer.h:26,
                 from src/camotics/gcode/Parser.cpp:23:
/builddir/build/BUILD/cbang-a4e236e8db124a02d73f8a5bca1e82ad44a40b6c/src/cbang/parse/Tokenizer.h: In instantiation of 'cb::Tokenizer<ENUM_T>::Tokenizer(cb::Scanner&) [with ENUM_T = cb::Enumeration<CAMotics::TokenTypeEnumeration>]':src/camotics/gcode/Tokenizer.h:34:71:   required from here
/builddir/build/BUILD/cbang-a4e236e8db124a02d73f8a5bca1e82ad44a40b6c/src/cbang/parse/Tokenizer.h:53:38: error: no matching function for call to 'cb::SmartPointer<cb::Scanner, cb::DeallocPhony, cb::RefCounterPhonyImpl>::SmartPointer(cb::Scanner&)'
   53 |       scanner(SmartPointer<Scanner>::Phony(scanner)) {}
      |                                      ^~~~~~~~~~~~~~
In file included from /builddir/build/BUILD/cbang-a4e236e8db124a02d73f8a5bca1e82ad44a40b6c/src/cbang/Exception.h:36,
                 from /builddir/build/BUILD/cbang-a4e236e8db124a02d73f8a5bca1e82ad44a40b6c/src/cbang/SmartPointer.h:33,
                 from src/camotics/gcode/ast/Block.h:24,
                 from src/camotics/gcode/Processor.h:24,
                 from src/camotics/gcode/Parser.h:24,
                 from src/camotics/gcode/Parser.cpp:21:
/builddir/build/BUILD/cbang-a4e236e8db124a02d73f8a5bca1e82ad44a40b6c/src/cbang/SmartPointer.h:143:5: note: candidate: 'cb::SmartPointer<T, DeallocT, CounterT>::SmartPointer(T*, cb::RefCounter*) [with T = cb::Scanner; DeallocT = cb::DeallocPhony; CounterT = cb::RefCounterPhonyImpl]'
  143 |     SmartPointer(T *ptr = 0, RefCounter *refCounter = 0) :
      |     ^~~~~~~~~~~~
/builddir/build/BUILD/cbang-a4e236e8db124a02d73f8a5bca1e82ad44a40b6c/src/cbang/SmartPointer.h:143:21: note:   no known conversion for argument 1 from 'cb::Scanner' to 'cb::Scanner*'
  143 |     SmartPointer(T *ptr = 0, RefCounter *refCounter = 0) :
      |                  ~~~^~~~~~~
/builddir/build/BUILD/cbang-a4e236e8db124a02d73f8a5bca1e82ad44a40b6c/src/cbang/SmartPointer.h:127:5: note: candidate: 'cb::SmartPointer<T, DeallocT, CounterT>::SmartPointer(const SmartPointerT&) [with T = cb::Scanner; DeallocT = cb::DeallocPhony; CounterT = cb::RefCounterPhonyImpl; cb::SmartPointer<T, DeallocT, CounterT>::SmartPointerT = cb::SmartPointer<cb::Scanner, cb::DeallocPhony, cb::RefCounterPhonyImpl>]'
  127 |     SmartPointer(const SmartPointerT &smartPtr) : refCounter(0), ptr(0) {
      |     ^~~~~~~~~~~~
/builddir/build/BUILD/cbang-a4e236e8db124a02d73f8a5bca1e82ad44a40b6c/src/cbang/SmartPointer.h:127:39: note:   no known conversion for argument 1 from 'cb::Scanner' to 'const SmartPointerT&' {aka 'const cb::SmartPointer<cb::Scanner, cb::DeallocPhony, cb::RefCounterPhonyImpl>&'}
  127 |     SmartPointer(const SmartPointerT &smartPtr) : refCounter(0), ptr(0) {
      |                  ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~

```

jcoffland, 2019-02-04:
Yes, there are many new changes to cbang.  You should lock the CAMotics 1.1.1 build to cbang tag 1.3.0.  I'm getting ready to make a new official CAMotics release.

jcoffland, 2019-02-04:
Sorry, yes.  v1.2.0 should work for CAMotics v1.1.1.

closed 2019-02-05 by jcoffland