CauldronDevelopmentLLC/CAMotics #285: 3D rendering with OpenGL ES 2.0

jcoffland, 2019-02-06:
This would make CAMotics more compatible with systems that do not have proper OpenGL drivers.  For example, Qt5 on Windows has ANGLE which automatically translates OpenGL ES 2.0 to DirectX.  OpenGL ES 2.0 would also make it easier to use textures.

See examples of Phong shading using ES 2.0:

 * http://www.cs.toronto.edu/~jacobson/phong-demo/
 * http://www.mathematik.uni-marburg.de/~thormae/lectures/graphics1/code/WebGLShaderLightMat/ShaderLightMat.html

renato-grottesi, 2022-06-22:
You can even make some procedural wood textures to allow people to try different colors :-)
https://godotshaders.com/shader/procedural-grain-wood-shader/
And probably some screen space ambient occlusion would help a lot to add some depth to the rendering of the carvings, considering that there can be a lot of deep cuts.
https://learnopengl.com/Advanced-Lighting/SSAO

jcoffland, 2022-06-26:
Good idea.

closed 2023-08-10 by jcoffland