CauldronDevelopmentLLC/CAMotics #290: No image in Simulation View

kg5000, 2019-02-12:
I just upgraded to the latest Camotics v1.20. When I load and simulate a GCode file I see nothing drawn in the SImulation View (black box). The file simulates fine in Vectric VCarve Desktop. I moved the position slider back and forth and tried different view modes (Wire surface, cut view, etc.).

I have been using Camotics for several years with no problems before. I even tested the app with a GCode file that worked on an older version of Camotics.

I am running Windows 10 64-bit Version 10.0.17134 Build 17134 on an HP Pavillion G6 laptop (Pentium Dual Core with Intel Integrated graphics).

[image]

jcoffland, 2019-02-13:
Does it work if you update your Intel graphics driver using this file:  https://downloadmirror.intel.com/24971/a08/win64_152824.exe

kg5000, 2019-02-16:
Unfortunately not. I tried that one and also the Intel driver helper app.
The app says my 2012 driver is the latest compatible version. The laptop
I'm running this on is pretty old. I have it running the latest Windows 10
but driver updates are an issue.

What version of Camotics could I downgrade to that would be compatible with
my old Intel HD Graphics? I forgot the Camotics version I was running
previously.

Thanks,
Ken

jcoffland, 2019-02-18:
There must be something different about your graphics drives.  CAMotics checks for OpenGL 2.1 support and will report an error and exit if it's not supported.  Since you are not getting that error, your system is saying that it does support OpenGL 2.1.  But still it's not working.

Are there any errors in the console?  You can open the console from the ``View`` menu.

Here are the last few versions of CAMotics for Windows:

 * [camotics_1.1.1_AMD64.exe](https://camotics.org/releases/public/release/camotics/windows-10-64bit/v1.1/camotics_1.1.1_AMD64.exe)
 * [camotics_1.1.0_AMD64.exe](https://camotics.org/releases/public/release/camotics/windows-10-64bit/v1.1/camotics_1.1.0_AMD64.exe)
 * [camotics_1.0.6_x86.exe](https://camotics.org/releases/public/release/camotics/windows-xp-32bit/v1.0/camotics_1.0.6_x86.exe)

jcoffland, 2019-02-18:
I've added a complete list of previous releases here: https://camotics.org/download.html#previous-releases

kg5000, 2019-02-18:
The console doesn't show much:

GCode::Tool Path bounds ((0,0,-8.89), (593.027,591.497,12.7)) dimensions
(593.027,591.497,21.59)

WARNING:Auto-creating missing tool 0

Computing surface bounded by ((-3.17809,-3.17809,-11.1378),
(638.178,638.178,3.00978)) at 2.35793 grid resolution

Time: 2.05 secs Triangles: 371744 Triangles/sec: 181431.91

kg5000, 2019-02-18:
I downgraded to v1.1.1 and it's working good. I'm good for now but if you
would like me to reinstall 1.2.0 or another future build to try it and
gather logs for this problem let me know. I am a software engineer and have
faced many a pesky problem I can't reproduce. It's good to have a system
where a problem can be reproduced.

Thanks for your help,
Ken

jcoffland, 2019-02-19:
There was a similar problem with previous versions of CAMotics and the Intel HD Graphics 4000.  This is documented in #159.  I'm still not sure what's going on here.  It's not the same problem as in #159.  

Thanks for the offer to retest.  Much appreciated.  I will take you up on this when I have something to test.

closed 2023-08-10 by jcoffland