CauldronDevelopmentLLC/CAMotics #296: Implement cutter radius compensation (G40, G41.1, G42, G42.1)

jcoffland, 2019-02-24:
http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g41-g42

Ian8555, 2022-02-21:
Any news or movement on this?

jcoffland, 2022-02-21:
No progress on this.  It's not too difficult to implement.  It just takes time/money that I've not had available.

Ian8555, 2022-02-23:
Thanks, I am struggling with trying to find another solution and going around in circles. How much would it cost if I were to pay for this to be done? Thanks

jcoffland, 2022-02-24:
I've looked again at what this would take.  I couldn't do it for less than $5k USD.  And that is to implement CRC as defined by LinuxCNC.

    http://linuxcnc.org/docs/2.4/html/gcode_tool_compensation.html#sec:Cutter-Radius-Compensation

Are you sure this will solve your problems?  Cutter radius compensation is fairly restrictive.  It only looks at the current and next move.  If you have any moves that are shorter than the radius of the cutter it may result in gauging.  If you need more complex polygon offsetting there are other ways to do this outside of GCode.

What is your input format?  Can you explain more what you are attempting to do and why you need cutter radius compensation?  There might be a better route.

Here is some more info on CRC from Mach: https://machmotion.com/blog/what-is-cutter-compensation/

OldManfromScene34, 2026-03-10:
Somewhere I have a Siemens 810 manual which explains the settings and ways tool compensation worked on it. All the G80 cycles, linear, arc moves, tool diameter compensation and such worked on all three coordinate planes perfectly. I learned to program by reading the commissioning manual and became a better programmer once I found the part explaining the G17/G18/G19 planes and tool comp to me. There were Numerous examples of when a non programmed move had to be inserted to achieve the commands directed by the GCode, such as making a small tangential move of less than the radius of the with comp on, moves that complete the profile programmed without an excess of cutting air, and tight inside or outside corner moves that interfere.  It would be a complete guide to programming your own motion controller as it list all the solutions to tool compensation in two dimensions.

It allows tool comp to be applied tangentially or to glide on or off of a part, which I never used as I had a TI calculator with an < button I used to enter a vector to approximate a position from a profile a guaranteed minimum distance from the desired tool path, just by guessing an angle and adding a small amount to the tool radius for the distance. I never saw the point of gliding onto a tool path unless the machine had backlash issues or someone wanted a very very pretty part finish, which i always used tangent radius moves onto and off of a part profile in that case.  

It's useful to see the graphical examples in the book to write your own tool comp engine, as Some of the solutions require more of a buffer than calculating the end of the current move based on the geometry of the next move.  The best tool comp engines look ahead several moves and also look at the path forwards and backwards.

If I find the book, I will send it to you. 

Erik Temesi