CauldronDevelopmentLLC/CAMotics #297: Is mysqlclient used in CAMotics?

hpmachining, 2019-03-09:
If I build CAMotics as is, I get a hard dependancy to libmariadb.so.3 (libmysqlclient.so). If I add my LDFLAGS, `"-Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now"`, to the LINKFLAGS in SConstruct I no longer get the dependancy to libmariadb, and the application appears to run properly without it installed. From what I can tell,  the `--as-needed` flag is removing it. So is mysqlclient used for any added functionality in CAMotics?

For reference, I am building in a [clean chroot](https://wiki.archlinux.org/index.php/DeveloperWiki:Building_in_a_clean_chroot) in Arch Linux using these packages:
[v8-3.14](https://aur.archlinux.org/packages/v8-3.14/)
[cbang](https://aur.archlinux.org/packages/cbang/)
[CAMotics](https://aur.archlinux.org/packages/camotics/)
Click the "View PKGBUILD" link to see the build scripts.

jcoffland, 2019-03-09:
mysql is not needed for CAMotics.  I did not know about the ``--as-needed`` option.  Nice trick.  Another way to handle this is to just not install the mysql dev packages in your chroot.

hpmachining, 2019-03-09:
Thanks for the info. I have mariadb-libs as a depends for cbang, so it gets automatically installed along with cbang when building CAMotics. Arch doesn't use separate dev packages. I should probably change the  depends in cbang to be an optional depends.

closed 2019-03-09 by hpmachining

jcoffland, 2019-03-09:
Are you making an official Arch package?

hpmachining, 2019-03-10:
Not official, but I'm maintaining the packages in the [Arch User Repository](https://aur.archlinux.org/) (AUR). The links for cbang and CAMotics in my comment above, plus a [cbang-git](https://aur.archlinux.org/packages/cbang-git/) package. I also intend to add a CAMotics-git package, as well. “Official” requires a TU (Trusted User) to adopt the package and add it to the official repository as a prebuilt binary package. AUR just provides the build script and any necessary patches and such for users to build a package themselves.

jcoffland, 2019-03-11:
Ok, good to know.