CauldronDevelopmentLLC/CAMotics #298: What even?

Leviathan220, 2019-03-22:
There is no cutter compensation. Even bigger of an issue is that the software doesn't put decimal points after each X, Y, Z. This makes the software completely unusable as I have to manually correct every line of G-Code the software prints out. Idk what file is used for translation, but if someone would let me know I'll fix it myself.

Leviathan220, 2019-03-22:
Seriously though I'll implement these things myself if someone can point me to the write file.

Leviathan220, 2019-03-23:
So to better describe the issue here. All feed rates coordinates, spindle speed, basically any value set by the user, MUST have a decimal point at the end. If they do not the machine will default to the slowest possible metric on the machine. So for instance I was using this program [output.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/2998204/output.txt)
. The machine will move to X0.0005 and Y0.0003 without a decimal point. I can manually add them in sure but at that point why am I using TPL to begin with. Also this is a copy of the file used to create the output of gcode minus the alterations I made afterwards(G20 instead of G21)(M8,M9) nothing major.

jcoffland, 2019-03-25:
CAMotics follows the GCode conventions used in LinuxCNC and vicariously RS274/NGC.  In section 3.3.2.1 of the RS274/NGC spec, it says:

```
A number consists of (1) an optional plus or minus sign, followed by (2) zero to many digits, followed, possibly, by (3) one decimal point, followed by (4) zero to many digits — provided that there is at least one digit somewhere in the number.
```

So trailing decimal points are *not* required.  What machine are you using that requires this?

The lack of cutter compensation support is documented in #296.

Leviathan220, 2019-03-26:
This would be a Haus VF-3 possibly a different model but I believe all Haus machines work this way. I guess I can write a bit of code to just cycle through and add a decimal point but it would be better if I could just create something to build into camotics. Well really all I use use is the cmd tools to convert my .tpl files into usable .nc files for the Haus machines we use where I work, or at least that was the idea. I really like the camotics software btw. The Haus has a built in simulator which I use before running anything anyway to check for errors, so I rarely use camotics but if I were to do this at home I definitely would want to pick up a buildbotics kit to play with.

Leviathan220, 2019-03-26:
Like basically without tpl I would have to hand write hundreds of lines of G-Code for things. If it becomes too much then MasterCam would be used. I think tpl is a great middle ground though as you can easily write a reusable function to simple contours, or drilling, or litterally anything. Without the decimal point I'm afraid I wont have much of a use for it though. At least till I can afford to get my own machine based on buildbotics. :'(

Leviathan220, 2019-03-26:
Mazak machines also work this way. Just about every production CNC I can think of works this way. I have never used LinuxCNC. I have not met anyone who has ever used it. Is it just a toy used for hobbyist. Is it incompatible with other CNC machines. I have to have a decimal point though.

closed 2019-03-26 by jcoffland

Leviathan220, 2019-03-26:
Wonderful, this is great news. I have compiled from source and tested the application. I can say that it works flawlessly. I really appreciate this and will get to using it right away.