CauldronDevelopmentLLC/CAMotics #300: GCode editor screen not updating correctly

jcoffland, 2019-03-25:
Calin Anca wrote:

I am having some trouble trying to edit Gcode in the program console.  Basically what happens is, when i try to add code, edit code i can't even see the coursor unless i click on the simulation view tab and the go back to the project tab .i can for example use the backspace key to delete some lines in the code or add some new line and write code but it wont show up unless i click on the simulation view tab and then go back to the project.Hope you can help. I have attached a video with a recording of my screen while using the program so you can see what i mean.

My laptop runs windows 10 home its a Intel Core i5 5300HQ 8GB Ram Nvidia gtx 950m.

jcoffland, 2019-03-25:
It looks like the GCode tab is just not updating correctly.

I was not able to reproduce this problem so I'm not sure how to fix it.  Do you have the latest driver for your NVidia GPU installed?

closed 2023-08-10 by jcoffland

Zeannn, 2025-09-01:
This is happening to me on the latest version. I’m running on Windows 11 with Nvidia GPU. Anyone knows the fix to this?