CauldronDevelopmentLLC/CAMotics #301: Failed to load OpenGL 3D Graphics

mjknoll, 2019-03-31:
So I read the 2 other issues with opengl and I am receiving a similar error but I am using a 1060 with the newest Nvidia driver and still getting an error about opengl 2.0. I also got the opengl extentions viewer tool and see that it shows 3.0+.  Must be something 2.0 specific it needs?

[image]

[image]

jcoffland, 2019-03-31:
It's likely CAMotics is finding another ``opengl.dll`` file somewhere on your system that does not support 2.0.  Do a search for ``opengl.dll`` to see how many of them can be found.

mjknoll, 2019-03-31:
Nothing but these.
[image]

jcoffland, 2019-03-31:
You could try running CAMotics under http://dependencywalker.com/ to see which dlls it's actually loading.

mjknoll, 2019-03-31:
Not very familiar with that software but the only reference to opengl I am finding after loading and profiling is this.

versionFunctions: Not supported on OpenGL ES

jcoffland, 2019-03-31:
Does it work if you set the environment variable ``QT_OPENGL=desktop``?

mjknoll, 2019-03-31:
Yes it runs with it set to desktop, was set to angle.

jcoffland, 2019-03-31:
If I understand you correctly, ``QT_OPENGL=angle`` was already set on your system?

mjknoll, 2019-03-31:
Yes it was set to angle that seems to break camotics. I can put anything else in there, even invalid arguments and it will run fine. I am assuming it was set by another piece of software that is using qt opengl.

jcoffland, 2019-03-31:
Yes, I think you are right.  ``QT_OPENGL=angle`` tells Qt to use OpenGL ES2.0 for everything.  CAMotics is not yet compatible with OpenGL ES2.0.  However, I plan to convert it to ES2.0.

closed 2023-08-10 by jcoffland