CauldronDevelopmentLLC/CAMotics #302: missing-call-to-setgroups-before-setuid

srakitnican, 2019-04-03:
It seems something changed from the version 1.1.1 that produces the  
`missing-call-to-setgroups-before-setuid` rpmlint error message.

I believe the rPath is added by the qt5 SCons build script, but I believe I can patch when building to remove if it is not possible to remove here.

```
/var/lib/mock/fedora-rawhide-x86_64/result/camotics-1.2.0.scons3fixes-1.fc31.x86_64.rpm
camotics.x86_64: W: incoherent-version-in-changelog 1.2.0-1 ['1.2.0.scons3fixes-1.fc31', '1.2.0.scons3fixes-1']
camotics.x86_64: E: binary-or-shlib-defines-rpath /usr/bin/gcodetool ['/usr/lib']
camotics.x86_64: E: missing-call-to-setgroups-before-setuid /usr/bin/gcodetool
camotics.x86_64: E: binary-or-shlib-defines-rpath /usr/bin/camotics ['/usr/lib']
camotics.x86_64: E: missing-call-to-setgroups-before-setuid /usr/bin/camotics
camotics.x86_64: E: binary-or-shlib-defines-rpath /usr/bin/tplang ['/usr/lib']
camotics.x86_64: E: missing-call-to-setgroups-before-setuid /usr/bin/tplang
camotics.x86_64: E: binary-or-shlib-defines-rpath /usr/bin/planner ['/usr/lib']
camotics.x86_64: E: missing-call-to-setgroups-before-setuid /usr/bin/planner
camotics.x86_64: E: binary-or-shlib-defines-rpath /usr/bin/camsim ['/usr/lib']
camotics.x86_64: E: missing-call-to-setgroups-before-setuid /usr/bin/camsim
camotics.x86_64: W: only-non-binary-in-usr-lib
camotics.x86_64: W: no-manual-page-for-binary gcodetool
camotics.x86_64: W: no-manual-page-for-binary tplang
camotics.x86_64: W: no-manual-page-for-binary planner
camotics.x86_64: W: no-manual-page-for-binary camotics
camotics.x86_64: W: no-manual-page-for-binary camsim
1 packages and 0 specfiles checked; 10 errors, 7 warnings.
```

srakitnican, 2019-04-03:
```
missing-call-to-setgroups-before-setuid:
This executable is calling setuid and setgid without setgroups or initgroups.
There is a high probability this means it didn't relinquish all groups, and
this would be a potential security issue to be fixed. Seek POS36-C on the web
for details about the problem.

```

jcoffland, 2019-04-03:
The *missing-call-to-setgroups-before-setuid* rpmlint error actually has nothing to do with CAMotics.  The code that calls ``setuid()`` is not reachable from CAMotics.  I made a change in cbang to use ``setgroups()`` instead of ``setgid()``.  Hopefully this fixes it.

srakitnican, 2019-04-04:
The change satisfies rpmlint, thanks.

closed 2019-04-08 by jcoffland