CauldronDevelopmentLLC/CAMotics #308: debian 10, camotics 1.2.0

cnczzz, 2019-08-12:
Does not work. At startup, it produces: 

`Opening /usr/share/doc/camotics/examples/camotics/camotics.camotics
/usr/bin/tplang: symbol lookup error: /usr/bin/tplang: undefined symbol: _ZN2v811HandleScopeC1Ev`

Console output:

*********************************** CAMotics ***********************************
    Version: 1.2.0
     Author: Joseph Coffland <joseph@cauldrondevelopment.com>
        Org: Cauldron Development LLC
  Copyright: 2011-2019, Joseph Coffland
   Homepage: https://camotics.org/
    License: https://www.gnu.org/licenses/gpl-2.0.txt
       Date: Feb 9 2019
       Time: 03:47:10
   Revision: 0554b96aa8ba78e55ff1d4bf6442cc04081e083c
     Branch: master
   Compiler: GNU 6.3.0 20170516
    Options: -std=c++11 -Wno-deprecated-declarations -O3 -funroll-loops -fno-pie
             -fPIC
   Platform: linux2 4.19.0-1-amd64
       Bits: 64
       Mode: Release
************************************ CBang *************************************
       Date: Feb 9 2019
       Time: 00:22:22
   Revision: e231d612a6fbb3424b7b211d08bb367f6d9ca0f8
     Branch: master
   Compiler: GNU 6.3.0 20170516
    Options: -std=c++11 -O3 -funroll-loops -fno-pie -fPIC
   Platform: linux2 4.19.0-1-amd64
       Bits: 64
       Mode: Release
************************************ System ************************************
        CPU: Intel(R) Pentium(R) CPU 2117U @ 1.80GHz
     CPU ID: GenuineIntel Family 6 Model 58 Stepping 9
       CPUs: 2
     Memory: 7.69GiB
Free Memory: 4.73GiB
    Threads: POSIX_THREADS
 OS Version: 4.19
Has Battery: true
 On Battery: false
 UTC Offset: 3
        PID: 10242
        CWD: /home/me
        DPI: 96
********************************************************************************
Opening /usr/share/doc/camotics/examples/camotics/camotics.camotics
/usr/bin/tplang: symbol lookup error: /usr/bin/tplang: undefined symbol: _ZN2v811HandleScopeC1Ev

Help!

jcoffland, 2019-08-12:
That looks like an incompatibility with the ``libv8.so`` on your system.  Did you do anything special to get CAMotics to link to v8?  What's the output of this command?:

    ldd /usr/bin/tplang

I've been looking into compiling v8 directly in to CAMotics to avoid this kind of problem.  But it would be even better if someone would help me get CAMotics accepted as a Debian package.

cnczzz, 2019-08-13:
Yes, I did link to libv8. Without it, camotics doesn't start. Here is the ldd output:
`	linux-vdso.so.1 (0x00007ffc6f9d2000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f6e49485000)
	libv8.so.3.14.5 => /lib/x86_64-linux-gnu/libv8.so.3.14.5 (0x00007f6fba361000)
	libssl.so.1.1 => /lib/x86_64-linux-gnu/libssl.so.1.1 (0x00007f6e493f4000)
	libcrypto.so.1.1 => /lib/x86_64-linux-gnu/libcrypto.so.1.1 (0x00007f6e4910a000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f6e49105000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f6e490e4000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f6e48f5f000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f6e48f45000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f6e48d84000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f6e49620000)
`

Did I do the wrong thing?

cnczzz, 2019-08-13:
Yes, it was libv8. Just downloaded libv8.so.3.4.15 from the Internet, deleted the link and copied to /usr/lib/x86_64-linux-gnu/. Now work. Thanks for your help.

jcoffland, 2019-08-13:
Great!

closed 2019-08-13 by jcoffland