CauldronDevelopmentLLC/CAMotics #309: Debian buster : cannot compile and cannot execute precompiled

sebmay, 2019-08-14:
root@buster:~/CAMotics# scons
scons: Reading SConscript files ...
EnvironmentError: No module named six:
  File "/root/CAMotics/SConstruct", line 37:     env.Tool('qt5', toolpath = ['./config'])
  File "/usr/lib/scons/SCons/Environment.py", line 1788:     tool = SCons.Tool.Tool(tool, toolpath, **kw)
  File "/usr/lib/scons/SCons/Tool/__init__.py", line 118:     module = self._tool_module()
  File "/usr/lib/scons/SCons/Tool/__init__.py", line 153:     raise SCons.Errors.EnvironmentError(e)

AND

root@buster:~# camotics
camotics: error while loading shared libraries: libv8.so.3.14.5: cannot open shared object file: No such file or directory
 ** is logic, i really need to recompîle camotics ****

Can you help please ?

jcoffland, 2019-08-14:
To compile try adding these packages:

    sudo apt-get install -y python3-six python-six libnode-dev

You should be able to install libv8 with:

    sudo apt-get install -y libv8-3.14.5

However, I've been having a problem with building against libv8-3.14.5 on Debian.  Recently Debian removed the libv8-dev package and replaced it with libnode-dev.  libnode-dev includes libv8 but a new version with a very different interface.  So recent version of CAMotics (what's on Github) require this newer version of libv8.  It's a mess.

I'm thinking about including a version of the libv8 code directly in the CAMotics source tree and statically linking it into the executable to avoid this problem altogether.

fra589, 2019-12-01:
Hi @jcoffland,  
What's news about this?  
I recently reinstall my main CNC + Dev computer with Debian Bullseye, and theresult is Camotics does not work anymore after reinstalled  the camotics_1.2.0_amd64.deb package. :-(  
I installed libnode-dev in place of libv8* but it doesn't work better.  
What can I do to have a functional Camotics again?  
T.I.A.  
@++;  
Gauthier.

fra589, 2019-12-04:
Hi all,
Sorry for the noise,
Finally, I succeeded to compile Camotics on my new Bullseye Debian installation.
Thanks for this great software.
@++;
Gauthier.

jcoffland, 2020-03-29:
You can also try this:

    sudo apt-get update
    sudo apt-get install -y wget
    wget http://ftp.us.debian.org/debian/pool/main/libv/libv8-3.14/libv8-3.14.5_3.14.5.8-8.1_amd64.deb
    wget http://ftp.us.debian.org/debian/pool/main/libv/libv8-3.14/libv8-3.14-dev_3.14.5.8-8.1_amd64.deb
    sudo dpkg -i libv8-3.14.5_3.14.5.8-8.1_amd64.deb libv8-3.14-dev_3.14.5.8-8.1_amd64.deb

djcurnow, 2020-03-30:
I tried the above but I get this from the terminal. Don't know how to get around this.
$ sudo dpkg -i libv8-3.14.5_3.14.5.8-8.1_amd64.deb libv8-3.14-dev_3.14.5.8-8.1_amd64.deb
[sudo] password for curnow: 
(Reading database ... 169395 files and directories currently installed.)
Preparing to unpack libv8-3.14.5_3.14.5.8-8.1_amd64.deb ...
Unpacking libv8-3.14.5 (3.14.5.8-8.1) over (3.14.5.8-8.1) ...
dpkg: regarding libv8-3.14-dev_3.14.5.8-8.1_amd64.deb containing libv8-3.14-dev:
 libv8-3.14-dev conflicts with libv8-dev
  libnode-dev:amd64 provides libv8-dev and is present and installed.

dpkg: error processing archive libv8-3.14-dev_3.14.5.8-8.1_amd64.deb (--install):
 conflicting packages - not installing libv8-3.14-dev
Setting up libv8-3.14.5 (3.14.5.8-8.1) ...
Processing triggers for libc-bin (2.30-0ubuntu2) ...
Errors were encountered while processing:
 libv8-3.14-dev_3.14.5.8-8.1_amd64.deb

jcoffland, 2020-03-30:
Run this first:

    sudo apt-get remove -y libnode-dev

curnowdj266, 2020-03-30:
OK did the remove. I uninstalled camotics and reinstalled.it. I also shut down and restarted the computer and I get the terminal message below when I try to run camotics.
$ camotics
camotics: error while loading shared libraries: libQt5WebSockets.so.5: cannot open shared object file: No such file or directory

jcoffland, 2020-03-30:
According to #322 that should be solved with this:

    sudo apt-get install libqt5websockets5-dev

closed 2023-08-10 by jcoffland