CauldronDevelopmentLLC/CAMotics #313: compilation error: ‘QOpenGLContext’ does not name a type

koendv, 2019-09-28:
I'm trying to compile cbang + camotics on raspberry pi 4 running Debian Buster. 
I'm using the following install:
`
sudo apt-get -y install scons build-essential libssl-dev qt5-default libqt5websockets5-dev libqt5opengl5-dev libv8-dev git libglu1-mesa-dev libyaml-dev libevent-dev libre2-dev libnode-dev libdxflib-dev
git clone https://github.com/CauldronDevelopmentLLC/cbang
git clone https://github.com/CauldronDevelopmentLLC/CAMotics
export TARGET_ARCH=arm-linux-gnueabihf
scons -C cbang
export CBANG_HOME=$PWD/cbang
cd CAMotics
scons
`
[buildlog.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/3665362/buildlog.txt)
Cbang compiles cleanly, but in camotics I get the following error: 
`src/camotics/view/GL.h:37:3: error: ‘QOpenGLContext’ does not name a type; did you mean ‘QMessageLogContext’?
   QOpenGLContext &getGLCtx();
   ^~~~~~~~~~~~~~
   QMessageLogContext
src/camotics/view/GL.h:39:11: error: ‘QOpenGLFunctions_2_0’ does not name a type
   typedef QOpenGLFunctions_2_0 GLFuncs;
	   ^~~~~~~~~~~~~~~~~~~~
src/camotics/view/GL.h:40:3: error: ‘GLFuncs’ does not name a type
   GLFuncs &getGLFuncs();
   ^~~~~~~
`

I attach a build log:
[buildlog.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/3665382/buildlog.txt)
Any suggestions?

jcoffland, 2019-09-28:
Looks like your Qt does not support OpenGL 2.0.  It might only support OpenGL ES.

Even if you got CAMotics built for the RaspberryPi it would be *really* slow.  The RPi does not have enough power.  Several other people have tried.  I don't think anyone has succeeded.  But don't let that stop you. ;)

koendv, 2019-10-01:
OK. Thank you for pointing me in the right direction.

Summary: Compiles. Runs. Simulates. The speed is sufficient for my purposes.

[image]

closed 2019-10-01 by koendv

jcoffland, 2019-10-01:
Nice.

koendv, 2019-10-02:
FYI: I've put up a binary package, together with build notes, on [github](https://github.com/koendv/camotics-raspberrypi), and posted an announcement on the raspberry pi [forum](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=253360).
The build notes might be of interest; had to modify SConfig a bit.

jcoffland, 2019-10-02:
Thanks for this. You might be interested in how I build part of CAMotics for the Buildbotics CNC controller using a Debian chroot environment.  It's cleaner because you don't have to modify your system.  Although it's slower since it emulates ARM using qemu rather than cross-compiling.  It would be best to combine both options.

The process starts with ``make gplan`` which runs these commands:

	./scripts/gplan-init-build.sh
	git -C rpi-share/cbang fetch
	git -C rpi-share/cbang reset --hard FETCH_HEAD
	git -C rpi-share/camotics fetch
	git -C rpi-share/camotics reset --hard FETCH_HEAD
	cp ./scripts/gplan-build.sh rpi-share/
	sudo ./scripts/rpi-chroot.sh $(GPLAN_IMG) /mnt/host/gplan-build.sh

Above called scripts are all in https://github.com/buildbotics/bbctrl-firmware/blob/master/scripts/  rpi-chroot.sh does a chroot in to a real raspbian distro.  It could be Debian with a cross compiler instead.  This would make updating your package much easier.