CauldronDevelopmentLLC/CAMotics #316: Prepare camotics for building AppImage

koendv, 2019-11-29:
Hi. AppImage is a way to distribute software as a single file. Inside the appimage there's a compressed filesystem, which contains not only the executeable, but also all libraries and resource files needed. At runtime, this filesystem gets mounted under /tmp. Your binary might be at /tmp/.mount12345/usr/bin/camotics, with some files at /tmp/.mount12345/usr/share/camotics, and others at /tmp/.mount12345/usr/share/doc/camotics. Unfortunately, camotics keeps looking for files in /usr/share/ . 

I've been looking at how prusa-slicer handles this. prusa-slicer looks for a resource directory relative to its binary, like this: (in PrusaSlicer.cpp)

```
boost::filesystem::path path_to_binary = boost::filesystem::system_complete(argv[0]);
boost::filesystem::path path_resources = path_to_binary.parent_path() / "../resources";
```
Just FYI.

jcoffland, 2019-11-30:
Done.

closed 2019-11-30 by jcoffland

koendv, 2019-12-27:
Thanks. Built. 
On x86 you might want to look at linuxdeployqt.

jcoffland, 2020-01-09:
Thanks.  I'm already using the macOS equivalent.

commit 24e37803 (2019-11-30): Relative path prefix.  Closes #316