CauldronDevelopmentLLC/CAMotics #317: Set tool origin/start point to workpiece center at toolpath's max Z

kfmut, 2020-01-10:
Hi,

I'm pretty sure that this question was asked before but I'm having troubles to find related "issues", so...

Why not to set default tool origin/start point to workpiece center at toolpath's max Z? It will  allow to remove first moves right thru workpiece if part origin is set at bottom face of workpiece.

[image]

[test_project.zip](https://github.com/CauldronDevelopmentLLC/CAMotics/files/4047760/test_project.zip)

jcoffland, 2020-01-10:
Usually, the way to fix this is to have the tool move to a safe z-height at the start and end of program.  Wouldn't the real CNC have this same problem?  If the simulator automatically moved the tool it would not catch this type of error.

closed 2020-01-10 by jcoffland

kfmut, 2020-01-10:
@jcoffland 

Sometimes it's done with G28 G91 Z0 move, but Camotics have no idea about G28 safe position so I think it simply skips it.

`N00001 G17
N00002 G21
N00003 G94
N00004 G28 G91 Z0
N00005 G90

N00006 T1 M6

...

N00046 G0 X53.506 Y11.025 S12000 M3
N00047 Z18.000
N00048 Z13.000
N00049 G1 Z9.000 F456
N00050 X53.299 Y11.348 Z8.954
N00051 X52.425 Y11.731 Z8.838`

jcoffland, 2020-01-10:
CAMotics does understand G28 but the position has to be set first.  G28 defaults to zero.  Unfortunately, there is currently no way to set this before the simulation starts.  I'm working on adding a machine configuration that could allow you to set up the initial state of the machine.

kfmut, 2020-01-12:
Yep, detailed machine configurations would be nice addition.