CauldronDevelopmentLLC/CAMotics #320: Show Cut Surface not working on High or Very High resolution settings

StevenMarston, 2020-02-07:
GCode::Tool Path bounds ((0,0,-0.127), (85.4862,25.4406,12.7)) dimensions (85.4862,25.4406,12.827)
Computing surface bounded by ((-2.20181,3.04875,-2.3413), (92.2575,29.0924,0.107949)) at 0.0816416 grid resolution
ERROR:std::exception: bad allocation

Works fine for medium and below with no error messages but doesn't yield the detail that I need.

I tried to look for a way in the forums for search for previous similar issues but never figured that part out.

[close your eyes.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/4171488/close.your.eyes.txt)

StevenMarston, 2020-02-07:
I should've said the previous issue is on the Version 1.2.0 32 bit version. It runs and renders fine on the Version 1.2.0 64 bit. Might help a little bit having that piece of information.

jcoffland, 2020-02-07:
``ERROR:std::exception: bad allocation`` means it ran out of memory.  At higher resolutions the simulation can require very large amounts of memory.

StevenMarston, 2020-02-07:
Thanks for the quick response!
Yeah, that makes sense.
Guess it's time to upgrade.
Works fine on the shop's PC which has more memory.
Appreciate you!

Regards,
Steven