CauldronDevelopmentLLC/CAMotics #323: Appimage for portable install

docop, 2020-04-10:
Hi  I wonder if the soft could have an appimage package as to deploy soo easily on every system ? Like on centos7.. Or could be a Snaps, package. This could be very great.

Thanks in advance, great soft.

jcoffland, 2020-04-10:
That would be great.  Care to submit a PR?

luzpaz, 2020-12-16:
@probonopd would you have any free time :rofl: to help create an appimage for this?

jcoffland, 2020-12-16:
Not at the moment but I could accept a PR.

luzpaz, 2020-12-17:
@jcoffland no worries, I was asking probonopd who's the person we can thank for designing and implementing AppImages in the first place. Much respect.

Harvie, 2021-02-27:
Yes please! I absolutely need AppImage for CAMotics! That would be great.

probonopd, 2021-02-27:
Sorry I don't have the time to do this at the moment but by following https://github.com/probonopd/linuxdeployqt#using-linuxdeployqt-with-travis-ci you should be able to get started. Let me know if you are running into any issues.

boris-so, 2022-10-26:
I found a recipe to create an appimage for camotics [here](https://github.com/AppImageCommunity/pkg2appimage/blob/master/recipes/CAMotics.yml), but that one did not worked on my debian 11(
I am not very experienced in process of creating appimages, but made a few modifications to that recipe and ended up with working one:
```yaml
app: CAMotics

ingredients:
  
  dist: bullseye
  sources:
    - deb http://ftp.gr.debian.org/debian/ bullseye main contrib non-free
  
  packages:
    - libnode-dev
    - camotics

  script:
    - wget -c https://camotics.org/builds/release/debian-stable-64bit/camotics_1.2.0_amd64.deb

script:
  # Old version of this library is required so rename the recent to look as old one. It seems to be compatible yet...
  - mv ./usr/lib/x86_64-linux-gnu/libv8.so ./usr/lib/x86_64-linux-gnu/libv8.so.3.14.5
```
Maybe someone else will have success with that :)