CauldronDevelopmentLLC/CAMotics #329: Subroutines

2EyeTec, 2020-07-27:
I'm trying to call subroutines, but Camotics is complaining, it can't find any. If I load the program in LinuxCNC (with the subroutines in the specified dir) it's working. I also tried to rename the subroutine from subroutine.ngc to subroutine (like mentioned in another thread). It says "Environment variable GCODE_SCRIPT_PATH not set", so I did a 

  `export GCODE_SCRIPT_PATH=/path/to/subroutines/`

But it's still not finding subroutines?! How can I load subroutines?

jcoffland, 2020-07-27:
You have to set ``GCODE_SCRIPT_PATH`` in the environment CAMotics runs in.  You can do so like this:

```
export GCODE_SCRIPT_PATH=/path/to/subroutines/
camotics
```

The subroutine name and file name must be the same.

2EyeTec, 2020-07-28:
Thanks! I got it working.

Regarding the name of the subroutine:

AFAIK I can't use o<subroutine.ngc> in my main program (the . is producing an error anyway), but LinuxCNC expects the subroutine name as subroutine.ngc, it can't find subroutine without the ngc. Don't you think LinuxCNC and Camotics should follow the same naming conventions?

jcoffland, 2020-07-29:
Yes, I agree.  I've modified the code to look for files with or with out the ``.ngc`` extension.  It will be in the next release.  Thanks for the input.  https://github.com/CauldronDevelopmentLLC/CAMotics/commit/4e023c06fd0a4dacebd77710256f1f9f47490590