CauldronDevelopmentLLC/CAMotics #331: Building on Win32

mkmantis, 2020-09-10:
I am trying to follow /doc/Windows_Build.md
I have successfully compiled v8 and cbang, or at least I think I have.
Building and packaging of CAMotics repository goes with warnings on python side, but msvc manages to produce binaries.
However, launching new camotics.exe leads to a message:

OpenGL Error
---------------------------
Failed to load OpenGL 3D graphics!
CAMotics requires OpenGL ES 2.0 or newer.
You may need to upgrade your graphics driver.

Additionally, no icons are drawn in main menu.
Build logs have a lot of warnings "cl : Command line warning D9002 : ignoring unknown option '/Ob3'"
It seems, CAMotics/win32 is meant to be built with visual studio 2019, since that flag was introduced there.

@jcoffland, would you kindly update versions info in Windows_Build.md?

mkmantis, 2020-09-10:
Digging further to find out what prevented GL to initialize, I found these exceptions:

> No precision specified for (float)

and then this article about openGL ES shader precision:
https://stackoverflow.com/questions/28540290/why-it-is-necessary-to-set-precision-for-the-fragment-shader

Adding 

> precision mediump float;

 to phong.frag solved my graphics problems and I can CAMotics almost fine now, except for missing menu icons.

closed 2020-09-10 by mkmantis

jcoffland, 2020-09-10:
I added that to the code.  Thanks!