CauldronDevelopmentLLC/CAMotics #334: g10 not working properly when modifying current datum

elsolo5000-2, 2020-10-31:
i have multiple small pieces spaced by x3.6mm,
i jog to my first piece and set my work zero g10 l20 p1 x0 y0 z0
i execute the operations
then i move the cutter to the next piece by g0 x3.6 y0
and i reset my work zero by g10 g10 l20 p1 x0 y0 z0
then reexecute the operations...

the first offset work, but the others are getting back near to machine zero
it seems i can not change offset of current datum

elsolo5000-2, 2020-10-31:
i get a warning
Position after synchronized command unknown in simulator.

jcoffland, 2020-11-04:
Can you submit a brief GCode example which exhibits the error?

elsolo5000-2, 2020-11-06:
on my cnc (grbl1.1f)
this trace a rectangle, and point "139" inside

code 1:
g94
f800
s12000
t1
m6
m3
g21
g90

g53 g0 x100 y50 (jog to workpiece zero)
g10 l20 p1 x0 y0 z0 (offset datum 1)

;contour
g1 z-1.0 (penetrating)
g1 x50
g1 y20
g1 x0
g1 y0

g0 z1 (safe z)
m5
g0 x14 y12 (case origin)
g10 l20 p1 x0 y0 z1  (offset datum 1)

;1
g0	x	1.4	y	4.2
g0 z-1.0
g0	x	1.8	y	4.6
g0	x	2.2	y	5
g0	x	2.2	y	4.6
g0	x	2.2	y	4.2
g0	x	2.2	y	3.8
g0	x	2.2	y	3.4
g0	x	2.2	y	3
g0	x	2.2	y	2.6
g0	x	2.2	y	2.2
g0	x	2.2	y	1.8
g0	x	2.2	y	1.4
g0	x	2.2	y	1
g0	x	2.2	y	0.6
g0	x	2.2	y	0.2

g0 z1 (safe z)
g0 x3.6 y0.0 (case end)
g10 l20 x0 y0 (offset datum 1)

;3
g0	x	0.6	y	4.8
g0 z-1.0
g0	x	1	y	4.8
g0	x	1.4	y	4.8
g0	x	1.8	y	4.8
g0	x	2.2	y	4.8
g0	x	2.6	y	4.8
g0	x	3	y	4.8
g0	x	2.6	y	4.4
g0	x	2.2	y	4
g0	x	2.2	y	3.6
g0	x	1.8	y	3.2
g0	x	2.2	y	3.2
g0	x	2.6	y	2.8
g0	x	3	y	2.4
g0	x	3	y	2
g0	x	3	y	1.6
g0	x	3	y	1.2
g0	x	3	y	0.8
g0	x	2.6	y	0.4
g0	x	2.2	y	0
g0	x	1.8	y	0
g0	x	1.4	y	0
g0	x	1	y	0.4
g0	x	0.6	y	0.8

g0 z1 (safe z)
g0 x3.6 y0.0 (case end)
g10 l20 x0 y0 (offset datum 1)

;9
g0	x	0.6	y	0.8
g0 z-1.0
g0	x	1	y	0.4
g0	x	1.4	y	0
g0	x	1.8	y	0
g0	x	2.2	y	0
g0	x	2.6	y	0.4
g0	x	3	y	0.8
g0	x	3	y	1.2
g0	x	3	y	1.6
g0	x	3	y	2
g0	x	3	y	2.4
g0	x	3	y	2.8
g0	x	3	y	3.2
g0	x	3	y	3.6
g0	x	3	y	4
g0	x	2.6	y	4.4
g0	x	2.2	y	4.8
g0	x	1.8	y	4.8
g0	x	1.4	y	4.8
g0	x	1	y	4.4
g0	x	0.6	y	4
g0	x	0.6	y	3.6
g0	x	0.6	y	3.2
g0	x	0.6	y	2.8
g0	x	1	y	2.4
g0	x	1.4	y	2
g0	x	1.8	y	2
g0	x	2.2	y	2
g0	x	2.6	y	2.4

g0 z1 (safe z)

elsolo5000-2, 2020-11-06:
for this code to work on camotics, i must achange current datum before make g10
like this
g53 g10 l20 x0 y0

or this
g55
g10 l20 x0 y0
g54

jcoffland, 2020-11-11:
```Position after synchronized command unknown in simulator.`` shows up when you run a probing move.  CAMotics cannot know when the machine would find the switch.  It also incorrectly displays this message after a pause (M1) command. But I don't see any such commands in your GCode so this is probably unrelated.

jcoffland, 2020-11-11:
I'm still a bit confused.  When I run the above program I get this:

[image]

What are you expecting it to do?

elsolo5000-2, 2020-11-12:
Your result look good. Mine was different. I will try it on my pc to see if the problem still here and try to isolate the faulty portion of the code. I gave you a part of code but it seems the problem is in another part

elsolo5000-2, 2020-11-13:
[image]
here is my result for the code i gave

jcoffland, 2020-11-17:
I get the same problem with 1.2.0.  It looks like I already fixed it.  It should work correctly with the next release which should come out in the next month or two.

closed 2023-08-10 by jcoffland