CauldronDevelopmentLLC/CAMotics #335: tool table import from FreeCAD

snelweg, 2020-12-09:
I try to import my FreeCAD 0.19 tooltable in json format. I get no errors, but there are also tools loaded in camotics.

What am I doing wrong?

[tooltable.json.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/5667939/tooltable.json.txt)

jcoffland, 2020-12-09:
There is a problem in the current release of CAMotics.  When you load a tool table, it gets loaded buy the user interface does not update so it still shows the previous tool table.  Does it work if you load the tool table, save the project then close and reopen the project?

snelweg, 2020-12-09:
> There is a problem in the current release of CAMotics. When you load a tool table, it gets loaded buy the user interface does not update so it still shows the previous tool table. Does it work if you load the tool table, save the project then close and reopen the project?

Thanks for your quick response.
No, when I do the above there is still nothing filled in in the tooltable spot.

I think it might be a problem with the way the .json file is made by FreeCAD. If I compare a tooltable file made from within CAMotics with a FC one there are many dissimilarties.

I will attach the files I used for comparing here.
[CAMotics.tools.json.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/5668352/CAMotics.tools.json.txt)
[FreeCAD.tooltable.json.txt](https://github.com/CauldronDevelopmentLLC/CAMotics/files/5668353/FreeCAD.tooltable.json.txt)

jcoffland, 2020-12-09:
I should have looked more closely.  Those are not compatible.  I'm talking to the FreeCAD guys.  We'll try to work something out for the future.

snelweg, 2020-12-09:
Great!
Thanks for your awesome piece of software.
Have a great day.

snelweg, 2020-12-10:
> I'm talking to the FreeCAD guys. We'll try to work something out for the future.

Sliptonic is the main developer of the Path workbench (that's where the g-code comes from, a manufacturing workbench).
My best bet is that he is the one who can make changes to make the import work.

https://forum.freecadweb.org/memberlist.php?mode=viewprofile&u=708

sliptonic, 2020-12-13:
I pushed a branch to my repo here:  https://github.com/sliptonic/FreeCAD/tree/feature/camotics
This is an experimental integration through camsim.   It requires a couple things:
1) It tests to see if camotics is installed.  If it's found, you'll get another button on the toolbar.
2) Your project needs to be using the new toolbit system not the legacy tools.
3) Your tool controller has to have a vertical and horizontal feedrate.

To use it, select a job in the tree and press the new camsim button.  In the background, Path will build a camotics project including a tooltable for the tools in the job.  It'll build a stock object from the stock object in the job and it will post-process the job with your selected post-processor.  All of the output gets written to a temp directory.
Camsim is called and the project passed in.  The resulting .stl solid is then loaded into the FreeCAD document.

Currently it writes the files to the temp directory and then deletes everything after completing the simulation.  But you could change the output location and then load the camotics.project file into camotics to see the actual simulation.

jcoffland, 2020-12-14:
That's awesome!

luzpaz, 2021-03-27:
Related: https://github.com/FreeCAD/FreeCAD/pull/4672

snelweg, 2021-03-27:
Wow, that looks so fing good!
What a great way to extend both FOSS software, what a nice collaboration.
Hopefully this will be picked up by @realthunder soon, I still prefer his branch over the stable 'official' branch.

luzpaz, 2022-06-27:
Can this ticket be closed?

snelweg, 2022-06-27:
It does not work here, so it should not be closed

luzpaz, 2022-06-27:
@snelweg is it https://github.com/FreeCAD/FreeCAD/pull/7104 ?

luzpaz, 2022-07-10:
@sliptonic you recently posted a [YT clip](https://youtu.be/RFu-DQ04uZM?t=786) in which you demo'ed using camotics internally and externally with FreeCAD. Is this issue solved as of this video?

sliptonic, 2022-07-10:
The current FreeCAD/Camotics integration lets the user generate a Camotics project from a Path Job.  
That isn't quite the same thing as importing a FreeCAD tooltable as a Camotics tooltable and I'm not sure exactly what the goal of the feature request is.

snelweg, 2022-07-11:
> I'm not sure exactly what the goal of the feature request is

This was hidden in one of your responses:

> Path will build a camotics project including a tooltable for the tools in the job

If that works it is usable and very nice. But it would be even nicer to be able to export the new FC Path tooltable to a Camotics tooltable so I can use the same tools even when I am not using the Camotics project build in Path.

sliptonic, 2022-07-11:
https://github.com/FreeCAD/FreeCAD/pull/7168

sliptonic, 2022-07-17:
The linked PR has been merged.  FreeCAD can now export a Camotics tool table.  
That doesn't exactly match the description of this issue which is to import a FreeCAD tooltable but it effectively meets the need.  I will let bug maintainers decide what to do with the issue.  
If you want to write an import routine, the FreeCAD tooltable information is stored in json.  Let me know if you need anything.

snelweg, 2022-07-17:
Thanks, this will save a lot problems with wrong tools in CAMotics. 
Since this solves the problem of the import and export isue I will close this request.

closed 2022-07-17 by snelweg