CauldronDevelopmentLLC/CAMotics #340: Build issue using cbang built with latest V8

hpmachining, 2021-02-14:
I built cbang using V8 version 9.0.136. Here is the cbang build command I used:
```
scons cxxstd="c++14" disable_local="libevent sqlite3 re2 libyaml zlib bzip2 expat"
```
Here is my camotics build command:
```
scons cxxstd="c++14" linkflags="$LDFLAGS"
```
Here are the build errors:
```
g++ -o build/camotics.so -Wl,--as-needed -Wl,-s -Wl,-x -Wl,--gc-sections -no-pie -pthread -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now -shared -Wl,-rpath=/usr/lib build/python.os -L/home/paul/aur/maintain/cbang-git/src/cbang/lib -L. build/libCAMoticsPy.a build/libCAMotics.a build/libCAMoticsGUI.a build/libclipper.a build/libDXF.a build/libSTL.a build/libGCode.a -lstdc++ -lcbang -lcbang-boost -lv8 -lssl -lcrypto -llz4 -lmysqlclient -lexpat -lbz2 -lz -lcrypt -lpthread -ldl -lutil -lm -lpython3.9 build/dxflib/libdxflib.a
/usr/bin/ld: /home/paul/aur/maintain/cbang-git/src/cbang/lib/libcbang.a(JSImpl.o): in function `cb::gv8::JSImpl::init(int*, char**)':
JSImpl.cpp:(.text._ZN2cb3gv86JSImpl4initEPiPPc+0x89): undefined reference to `v8::platform::NewDefaultPlatform(int, v8::platform::IdleTaskSupport, v8::platform::InProcessStackDumping, std::unique_ptr<v8::TracingController, std::default_delete<v8::TracingController> >)'
collect2: error: ld returned 1 exit status
scons: *** [camsim] Error 1
/usr/bin/ld: /home/paul/aur/maintain/cbang-git/src/cbang/lib/libcbang.a(JSImpl.o): in function `cb::gv8::JSImpl::init(int*, char**)':
JSImpl.cpp:(.text._ZN2cb3gv86JSImpl4initEPiPPc+0x89): undefined reference to `v8::platform::NewDefaultPlatform(int, v8::platform::IdleTaskSupport, v8::platform::InProcessStackDumping, std::unique_ptr<v8::TracingController, std::default_delete<v8::TracingController> >)'
collect2: error: ld returned 1 exit status
scons: *** [camotics] Error 1
scons: building terminated because of errors.
```
Let me know if there is anything you want me to try or provide. Thanks!

jcoffland, 2021-02-14:
Looks like it's missing the v8 platform library.  Newer versions of v8 may require linking against ``libv8_libplatform.a``.  I recommend building the v8 ``libv8_monolith.a`` library.  If you rename it ``libv8.a`` cbang will link against it.  I could also change the cbang config to look for these other library names.

hpmachining, 2021-02-14:
Thank you. I want to get it to work with my current dynamic v8 build, because there is already a package in the Arch Linux AUR (v8-r) that is being maintained. I got it to work with the following hacks in `cbang`:
```
sed -i "20i\    conf.CBRequireLib('v8_libplatform')" config/v8/__init__.py
sed -i "37i\    env.CBDefine('V8_COMPRESS_POINTERS')" SConstruct
```
Is there a better way to accomplish this? Can I pass the define `V8_COMPRESS_POINTERS` in the `scons` command line instead of hacking the `SConstruct` file?

jcoffland, 2021-02-16:
I added similar code to cbang to solve this.  The latest commit should work now.

closed 2021-02-16 by jcoffland

dresco, 2021-05-12:
@jcoffland Just FYI, the cbang commit from 16th Feb caused issues for my local CAMotics build on Fedora 33 (packages v8-devel-8.4.371). 

```
#
# Fatal error in , line 0
# Embedder-vs-V8 build configuration mismatch. On embedder side pointer compression is ENABLED while on V8 side it's DISABLED.
#
```

I just needed to comment out the #define V8_COMPRESS_POINTERS line to get working again..

jcoffland, 2021-05-12:
Good to know.  This really depends on the libv8 you are linking to.  Perhaps I should add an option to C! to select either option.  Unfortunately, libv8 does not have a ``config.h`` that specifies which why the lib was compiled.