CauldronDevelopmentLLC/CAMotics #347: Problem building CAMotics with dependencies

tpimh, 2021-07-27:
I'm trying to build CAMotics 1.2.0 with C! 1.6.0 and [embedded-v8](https://github.com/CauldronDevelopmentLLC/embedded-v8).

I've run into multiple problems during the compilation:

- embedded-v8 compiles successfully, but I couldn't find a proper way to install it (there are no "install" targets in generated ninja files)
- C! is unable to locate V8, thus it's built without V8 support
- CAMotics build is not starting with the following error:
```
SConsEnvironmentError: No module named config:
  File "/run/build/camotics/SConstruct", line 15:
    env.Tool('config', toolpath = [cbang])
  File "/app/lib/scons/SCons/Environment.py", line 1798:
    tool = SCons.Tool.Tool(tool, toolpath, **kw)
  File "/app/lib/scons/SCons/Tool/__init__.py", line 122:
    module = self._tool_module()
  File "/app/lib/scons/SCons/Tool/__init__.py", line 221:
    raise SCons.Errors.SConsEnvironmentError(error_string)
```

I'm using KDE SDK 5.15, additionally installing SCons 3.1.1 and Python Six 1.11.0.

jcoffland, 2021-07-28:
1. There's no need to install embedded-v8
2. You have to set several environment variables for this to work correctly with C!  See below.
3. This error is caused by not having ``CBANG_HOME`` set correctly.

I've added a ``scripts/build`` to CAMotics which runs on Debian Linux.  It installs all the required packages, builds embedded-v8 and C! and setups the environment variables.  You can just clone CAMotics and run this script.

I also added a script ``embedded-v8/build.sh`` which ``scripts/build`` calls.  It builds embedded-v8 and then moves the library so that C! has an easier time finding it.

I tested all of this on a fresh install of Debian bullseye as root like this:

```
apt-get update
apt-get install -y git sudo
git clone https://github.com/CauldronDevelopmentLLC/CAMotics
./CAMotics/scripts/build
```

If you're on an OS other than Debian Linux you can look at these scripts to see what you need to do.  The main thing that's different for other OS is how you install the required packages.

tpimh, 2021-07-28:
Thank you very much for publishing your build scripts, this helps a lot. I didn't find any references to `V8_HOME` in the C! source, so not sure where exactly this variable is being used. Are you sure that the script is using embedded_v8 headers and not system headers provided by `libv8-dev` package?

In order for C! config to find embedded-v8 I had to set `V8_INCLUDE` environment variable to `$V8_HOME/include`, then I got a huge number of errors (maybe this discussion should be moved to C! issues?):

```
In file included from src/cbang/js/v8/ValueRef.h:35,
                 from src/cbang/js/v8/JSImpl.h:35,
                 from src/cbang/js/Javascript.cpp:40:
src/cbang/js/v8/Value.h: In member function ‘virtual bool cb::gv8::Value::toBoolean() const’:
src/cbang/js/v8/Value.h:87:75: error: no matching function for call to ‘v8::Value::BooleanValue()’
   87 |       bool toBoolean() const {assertDefined(); return value->BooleanValue();}
      |                                                                           ^
In file included from src/cbang/js/v8/V8.h:43,
                 from src/cbang/js/v8/Value.h:35,
                 from src/cbang/js/v8/ValueRef.h:35,
                 from src/cbang/js/v8/JSImpl.h:35,
                 from src/cbang/js/Javascript.cpp:40:
embedded-v8/include/v8.h:3055:8: note: candidate: ‘bool v8::Value::BooleanValue(v8::Isolate*) const’
 3055 |   bool BooleanValue(Isolate* isolate) const;
      |        ^~~~~~~~~~~~
embedded-v8/include/v8.h:3055:8: note:   candidate expects 1 argument, 0 provided
In file included from src/cbang/js/v8/ValueRef.h:35,
                 from src/cbang/js/v8/JSImpl.h:35,
                 from src/cbang/js/Javascript.cpp:40:
src/cbang/js/v8/Value.h: In member function ‘virtual double cb::gv8::Value::toNumber() const’:
src/cbang/js/v8/Value.h:93:75: error: no matching function for call to ‘v8::Value::NumberValue()’
   93 |       double toNumber() const {assertDefined(); return value->NumberValue();}
      |                                                                           ^
In file included from src/cbang/js/v8/V8.h:43,
                 from src/cbang/js/v8/Value.h:35,
                 from src/cbang/js/v8/ValueRef.h:35,
                 from src/cbang/js/v8/JSImpl.h:35,
                 from src/cbang/js/Javascript.cpp:40:
embedded-v8/include/v8.h:3058:39: note: candidate: ‘v8::Maybe<double> v8::Value::NumberValue(v8::Local<v8::Context>) const’
 3058 |   V8_WARN_UNUSED_RESULT Maybe<double> NumberValue(Local<Context> context) const;
      |                                       ^~~~~~~~~~~
embedded-v8/include/v8.h:3058:39: note:   candidate expects 1 argument, 0 provided
In file included from src/cbang/js/v8/ValueRef.h:35,
                 from src/cbang/js/v8/JSImpl.h:35,
                 from src/cbang/js/Javascript.cpp:40:
src/cbang/js/v8/Value.h: In member function ‘virtual int cb::gv8::Value::toInteger() const’:
src/cbang/js/v8/Value.h:94:74: error: no matching function for call to ‘v8::Value::IntegerValue()’
   94 |       int toInteger() const {assertDefined(); return value->IntegerValue();}
      |                                                                          ^
In file included from src/cbang/js/v8/V8.h:43,
                 from src/cbang/js/v8/Value.h:35,
                 from src/cbang/js/v8/ValueRef.h:35,
                 from src/cbang/js/v8/JSImpl.h:35,
                 from src/cbang/js/Javascript.cpp:40:
embedded-v8/include/v8.h:3060:40: note: candidate: ‘v8::Maybe<long int> v8::Value::IntegerValue(v8::Local<v8::Context>) const’
 3060 |   V8_WARN_UNUSED_RESULT Maybe<int64_t> IntegerValue(
      |                                        ^~~~~~~~~~~~
embedded-v8/include/v8.h:3060:40: note:   candidate expects 1 argument, 0 provided
In file included from src/cbang/js/v8/ValueRef.h:35,
                 from src/cbang/js/v8/JSImpl.h:35,
                 from src/cbang/js/Javascript.cpp:40:
src/cbang/js/v8/Value.h: In member function ‘int32_t cb::gv8::Value::toInt32() const’:
src/cbang/js/v8/Value.h:100:74: error: no matching function for call to ‘v8::Value::Int32Value()’
  100 |       int32_t toInt32() const {assertDefined(); return value->Int32Value();}
      |                                                                          ^
In file included from src/cbang/js/v8/V8.h:43,
                 from src/cbang/js/v8/Value.h:35,
                 from src/cbang/js/v8/ValueRef.h:35,
                 from src/cbang/js/v8/JSImpl.h:35,
                 from src/cbang/js/Javascript.cpp:40:
embedded-v8/include/v8.h:3066:40: note: candidate: ‘v8::Maybe<int> v8::Value::Int32Value(v8::Local<v8::Context>) const’
 3066 |   V8_WARN_UNUSED_RESULT Maybe<int32_t> Int32Value(Local<Context> context) const;
      |                                        ^~~~~~~~~~
embedded-v8/include/v8.h:3066:40: note:   candidate expects 1 argument, 0 provided
In file included from src/cbang/js/v8/ValueRef.h:35,
[... 520 lines cut ...]
      |                                               ^
In file included from src/cbang/js/v8/V8.h:43,
                 from src/cbang/js/v8/Value.h:35,
                 from src/cbang/js/v8/Value.cpp:33:
embedded-v8/include/v8.h:3019:44: note: candidate: ‘v8::MaybeLocal<v8::Object> v8::Value::ToObject(v8::Local<v8::Context>) const’
 3019 |   V8_WARN_UNUSED_RESULT MaybeLocal<Object> ToObject(
      |                                            ^~~~~~~~
[... cut ...]
embedded-v8/include/v8.h:3019:44: note:   candidate expects 1 argument, 0 provided
src/cbang/js/v8/JSImpl.cpp: In static member function ‘static void cb::gv8::JSImpl::init(int*, char**)’:
src/cbang/js/v8/JSImpl.cpp:74:44: error: ‘CreateDefaultPlatform’ is not a member of ‘v8::platform’; did you mean ‘NewDefaultPlatform’?
   74 |   v8::V8::InitializePlatform(v8::platform::CreateDefaultPlatform());
      |                                            ^~~~~~~~~~~~~~~~~~~~~
      |                                            NewDefaultPlatform
```

jcoffland, 2021-07-28:
You need to pull the latest version of C! yours is outdated.  ``V8_HOME`` is accessed through this code: https://github.com/CauldronDevelopmentLLC/cbang/blob/master/config/__init__.py#L46

Also setting ``V8_HOME`` is the same as setting ``V8_INCLUDE=$V8_HOME/include`` and ``V8_LIBPATH=$V8_HOME/lib``.  Those are the default paths.

I recommend uninstalling ``libv8-dev`` if you're using ``embedded-v8``.

tpimh, 2021-08-01:
@jcoffland Thank you very much!

Using C! source from the master branch instead of the latest release resolved the compilation problem (time for a new release, maybe?), however `SConsEnvironmentError: No module named config` problem still persists. Do you have any suggestions what my install may be lacking?

tpimh, 2021-08-02:
The problem was actually caused by C! source being removed from it original path. After restoring the source, the new error now looks like this (using SCons 4.2.0):

```
scons: Reading SConscript files ...
AttributeError: module 'SCons.Warnings' has no attribute 'Warning':
  File "/run/build/camotics/SConstruct", line 37:
    env.Tool('qt5', toolpath = ['./config'])
  File "/app/lib/python3.8/site-packages/SCons-4.2.0-py3.8.egg/SCons/Environment.py", line 1878:
    tool = SCons.Tool.Tool(tool, toolpath, **kwargs)
  File "/app/lib/python3.8/site-packages/SCons-4.2.0-py3.8.egg/SCons/Tool/__init__.py", line 120:
    module = self._tool_module()
  File "/app/lib/python3.8/site-packages/SCons-4.2.0-py3.8.egg/SCons/Tool/__init__.py", line 209:
    module = spec.loader.load_module(spec.name)
  File "<frozen importlib._bootstrap_external>", line 462:
    
  File "<frozen importlib._bootstrap_external>", line 962:
    
  File "<frozen importlib._bootstrap_external>", line 787:
    
  File "<frozen importlib._bootstrap>", line 265:
    
  File "<frozen importlib._bootstrap>", line 702:
    
  File "<frozen importlib._bootstrap>", line 671:
    
  File "<frozen importlib._bootstrap_external>", line 783:
    
  File "<frozen importlib._bootstrap>", line 219:
    
  File "/run/build/camotics/config/qt5/__init__.py", line 45:
    class ToolQt5Warning(SCons.Warnings.Warning): pass
```

jcoffland, 2021-08-05:
Right, when ever you see ``SConsEnvironmentError: No module named config`` it most likely means ``CBANG_HOME`` is not set correctly.

The ``SCons.Warnings.Warning`` issue has already been fixed in the latest CAMotics.  It is caused by a change in SCons.

tpimh, 2021-08-09:
Thank you very much for your assistance! After downgrading SCons I still faced the issues described in #270, but using the master branch everything went smoothly. The build succeeded!

Do you remember what version of SCons did you use in order to build CAMotics 1.2.0?

I have published my Flatpak manifest [here](https://github.com/tpimh/flatcam-flatpak) (related to #273). It's not finished yet, there are still a lot of improvements to be done. I will try to submit it to the [Flathub](https://flathub.org/) repository, but probably only after C! 1.6.1 or 1.7.0 release that wouldn't require patching to be used with [embedded-v8](https://github.com/CauldronDevelopmentLLC/embedded-v8).

closed 2021-08-09 by tpimh

jcoffland, 2021-08-13:
I'm using scons v4.0.1.  I tagged CAMotics ``v1.2.2-rc1`` and C! ``1.6.1``.