CauldronDevelopmentLLC/CAMotics #349: SCons doesn't install documentation by default

tpimh, 2021-08-16:
I was under the impression that `scons install` would install machines and examples under `${install_prefix}/share/doc/camotics`, but seems that it doesn't. Everything works fine if I copy these two directories to that destination manually, but I believe this should be automated somehow. Is there another SCons target that does this?

jcoffland, 2021-08-17:
I don't use the ``scons install`` target.  Instead, I build a package with ``scons pacakge`` then install the package.  It's much better to use you OS' package system.

tpimh, 2021-08-18:
In this case, I'm writing a script that would build such a package. On one hand, I can build a Debian package, then just unpack it into `install_prefix`, but that seems overcomplicating things a little bit.

jcoffland, 2021-08-19:
I updated the ``install`` target.  What kind of package are you building? Thanks for the report and let me know if there's anything else.

closed 2021-08-19 by jcoffland

tpimh, 2021-08-20:
Thank you very much for your patch, it fixed the problem! The package I'm building is Flatpak, the manifest [can be found here](https://github.com/tpimh/camotics-flatpak), and now it's fully functional!

commit d6df48ad (2021-08-19): Update install build target. close: #349