CauldronDevelopmentLLC/CAMotics #351: Building camotics with build script fails

code8buster, 2021-10-01:
I'm using the script in scripts/build to build camotics with embedded_v8 and cbang
Keep getting this error and warning:
scons: warning: QT5DIR variable is not defined, using moc executable as a hint (QT5DIR=/usr)
... (some stuff in here checking libraries/headers)
...
scons: *** [build/camotics/qt/moc_AboutDialog.cc] TypeError `_defines() missing 2 required positional arguments: 'target' and 'source'' trying to evaluate `${_defines(QT5_MOCDEFPREFIX, CPPDEFINES, QT5_MOCDEFSUFFIX, __env__)}'
scons: building terminated because of errors.

code8buster, 2021-10-24:
This is a problem with scons 4.2.0
Downgraded to scons 4.1.1 and the program compiled

dresco, 2021-11-08:
Hi, @jcoffland any thoughts on a workaround for this? Am facing the same issue on Fedora 35 (where 4.2.0 is the earliest available scons). Thanks!

code8buster, 2021-11-23:
Install it with pip3, as a workaround
pip3 install SCons==4.1.0.post1
https://pypi.org/project/SCons/#history

tirithen, 2022-03-10:
I'm facing similar problems on Arch Linux from the AUR packages camotics (https://aur.archlinux.org/packages/camotics#comment-849352) and camotics-git (https://aur.archlinux.org/packages/camotics-git#comment-835431).

Is related to a larger task to get it compiling with scons 4.3.0 (and possibly also 4.2.0)?

hpmachining, 2022-03-12:
Thanks to the link @jhdub23 provided I came up with this patch to build the latest commit with scons 4.3.0.
```
--- a/config/qt5/__init__.py
+++ b/config/qt5/__init__.py
@@ -763,7 +763,7 @@ def generate(env):
         QT5_MOCDEFPREFIX = '-D',
         QT5_MOCDEFSUFFIX = '',
         QT5_MOCDEFINES = '${_defines(QT5_MOCDEFPREFIX, CPPDEFINES, '
-        'QT5_MOCDEFSUFFIX, __env__)}',
+        'QT5_MOCDEFSUFFIX, __env__, TARGET, SOURCE)}',
         QT5_MOCCPPPATH = [],
         QT5_MOCINCFLAGS = '$( ${_concat(QT5_MOCINCPREFIX, QT5_MOCCPPPATH, '
         'INCSUFFIX, __env__, RDirs)} $)',
```

jcoffland, 2022-03-14:
I added a fix that should work for both newer and older SCons versions.

closed 2022-03-14 by jcoffland