CauldronDevelopmentLLC/CAMotics #352: Fedora - app becomes unresponsive after opening native file open or save dialog

dresco, 2021-11-24:
Hi, trying to understand an issue I'm facing running on Fedora (building from source).

Any file open/save operation that launches the native Gnome file dialogs causes the app to become unresponsive. If I run it with sudo, then it uses (what look like) the Qt dialogs & all works fine. 

Just launching & then clicking on save project is enough to reproduce this here. There are no errors logged from the application when this happens, but a couple of lines that may be relevant in the system logs;

```
Nov 24 10:37:24 laptop.localdomain gnome-shell[3793]: Invalid window geometry for xdg_surface@41. Ignoring for now, but this will result in client termination in the future.
Nov 24 10:37:33 laptop.localdomain xdg-desktop-por[1823890]: Failed to create foreign window for XID 1
Nov 24 10:37:33 laptop.localdomain xdg-desktop-por[1823890]: Failed to associate portal window with parent window x11:1
```
Wondering whether you have any tips for troubleshooting? Or is there a way to force use of the Qt dialogs for a normal user session?

dresco, 2021-11-24:
FWIW, I've bodged around it temporarily with this..

```
$ git diff
diff --git a/src/camotics/qt/FileDialog.cpp b/src/camotics/qt/FileDialog.cpp
index 4329c04..d1bd7fc 100644
--- a/src/camotics/qt/FileDialog.cpp
+++ b/src/camotics/qt/FileDialog.cpp
@@ -49,6 +49,8 @@ QString FileDialog::open(const QString &title, const QString &filters,
 
   QFileDialog dialog(&win, title, qPath, filters);
 
+  dialog.setOption(QFileDialog::DontUseNativeDialog);
+
   if (save || anyFile) dialog.setFileMode(QFileDialog::AnyFile);
   else dialog.setFileMode(QFileDialog::ExistingFile);
 
```

jcoffland, 2023-08-10:
Particularly macOS users don't like it when we use non-native dialogs.  This seems more like a Fedora problem to me.

closed 2023-08-10 by jcoffland