CauldronDevelopmentLLC/CAMotics #354: Standard library of TPL operations (cutting rectangular pockets, etc)

zzaacchh, 2021-11-30:
I am really enjoying learning to use TPL and the Camotics simulator.  To gain a bit more leverage, I'm considering writing a TPL module that includes some higher-level CAM operations for cutting pockets, boring holes, surfacing, etc.  I imagine, however, that this is something that others might have already written.  If so, I wonder if anyone would be willing to supply their code as a starting point?

jcoffland, 2021-11-30:
Awesome.  Have you used the built in DXF library?  It has support for offsetting.  This is very useful for CAM operations.  Some of the built-in examples use this.  We are still missing many higher level functions like the ones you a suggesting.  It would be awesome if you created these.  If you are willing to release them as Open-Source and they are sufficiently well tested I will included them in the next CAMotics release.

zzaacchh, 2021-11-30:
Sure, Joseph.  I'll plan to prototype a module and share for feedback.  I'm relatively new to machining, so I will likely benefit from some editorial help on the toolpaths I implement.   If the idea feels like it's working, I'd be more than happy to have it open sourced and included in CAMotics.

BTW, I haven't tried the DXF features of TPL, as I don't tend to work with applications that produces these file types.   I primarily use JSCad to build models, not a traditional CAD application :)

jcoffland, 2021-11-30:
The DXF library loads DXF files and then turns them into polygon data.  It then calls the offsetting code with this data.  You might want to look at how it uses the offsetting library called ``clipper``.  Clipper is very good at offsetting paths.  It handles many difficult cases.  You can use clipper with out DXF.

jcoffland, 2021-11-30:
There is an undocumented built-in function called ``offset()`` you can see it being used here: https://github.com/CauldronDevelopmentLLC/CAMotics/blob/master/tpl_lib/dxf/dxf.tpl#L744

zzaacchh, 2021-11-30:
Thank you for the link, Joseph.  That's quite helpful.  I didn't realize that the dxf feature was itself implemented as a  javascript/tpl module.  I'll review and look for opportunities to reuse libraries and mimic code style.